PX4/PX4-Autopilot #15609: Gazebo SITL multiple run script gives MAVLink connection only to two UAVs

KrisRadowski, 2020-08-25:
When I'm running a Gazebo SITL multiple run script with more than 2 UAVs, I can connect over MAVlink only with 2 drones, because I didn't get remote address for rest of them

**To Reproduce**
run Gazebo SITL multiple run script with more than 2 UAVs

**Expected behavior**
for every drone remote address is given

Jaeyoung-Lim, 2020-08-25:
@KrisRadowski The remote is your local host, so I think this was already given for the two vehicles already

Jaeyoung-Lim, 2020-08-25:
@KrisRadowski You can check how the UDP ports are incremented in https://github.com/PX4/Firmware/blob/master/ROMFS/px4fmu_common/init.d-posix/rcS#L112

which script exactly are you using?

Jaeyoung-Lim, 2020-08-25:
@KrisRadowski I noticed that gzserver crashes if you spawn multiple vehicles in gazebo 9.13.0 and gazebo 9.14.0

Have you tried using gazebo11? On gazebo11 the issue is no longer there

KrisRadowski, 2020-08-26:
> which script exactly are you using?

I'm using `gazebo_sitl_multiple_run.sh` from `Tools` directory

> Have you tried using gazebo11? On gazebo11 the issue is no longer there

I assume I need to uninstall old version, right?

Jaeyoung-Lim, 2020-08-26:
@KrisRadowski Depends on your system. You can update or remove/install.

KrisRadowski, 2020-08-26:
> @KrisRadowski Depends on your system. You can update or remove/install.

on gazebo11 running `gazebo_sitl_multiple_run.sh` throws errors on parsing iris xacro:
```[Err] [Model.cc:99] Error Code 23 Msg: FrameAttachedToGraph error, Non-LINK vertex with name [gps0_joint] is disconnected; it should have 1 outgoing edge in MODEL attached_to graph.
[Err] [Model.cc:99] Error Code 23 Msg: Graph has __model__ scope but sink vertex named [gps0_joint] does not have FrameType LINK when starting from vertex with name [gps0_joint].
[Err] [Model.cc:99] Error Code 26 Msg: PoseRelativeToGraph error, Non-MODEL vertex with name [gps0_joint] is disconnected; it should have 1 incoming edge in MODEL relative_to graph.
[Err] [Model.cc:99] Error Code 26 Msg: PoseRelativeToGraph frame with name [gps0_joint] is disconnected; its source vertex has name [gps0_joint], but its source name should be __model__.
```
what I should change in `iris.xacro`?

Jaeyoung-Lim, 2020-08-26:
@KrisRadowski You don't need to change anything

KrisRadowski, 2020-08-26:
ok, now I'm gonna look into rcS file and figure out for myself why it's works not like it should

Jaeyoung-Lim, 2020-08-26:
@KrisRadowski So what is the failure you are having?

KrisRadowski, 2020-08-29:
> @KrisRadowski So what is the failure you are having?

I'm connecting with three UAVs in MAVSDK (any model):
```
connectionResult[0] = dc.add_any_connection("udp://127.0.0.1:14540");
connectionResult[1] = dc.add_any_connection("udp://127.0.0.1:14541");
connectionResult[2] = dc.add_any_connection("udp://127.0.0.1:14542");//this one is faulty
```

and a faulty one throws errors:
```
[06:52:02|Error] No known remotes (udp_connection.cpp:133)
[06:52:02|Error] send fail (mavsdk_impl.cpp:147)
[06:52:02|Error] connection send error (520) (mavlink_commands.cpp:291)
```

he's not detected by QGC as well:
[image]

lappie, 2020-09-05:
Running in to the same issue, when I run `./Tools/gazebo_sitl_multiple_run.sh -n 2` everything works fine. As soon as I change `-n` to anything higher than 2, the drones start up but immediately disconnect. 

I tried running it with gazebo11, same result.

So to reproduce the issue: `./Tools/gazebo_sitl_multiple_run.sh -n 3`

Jaeyoung-Lim, 2020-09-05:
@stmoon Could you maybe have a look?

stmoon, 2020-09-06:
> @stmoon Could you maybe have a look?

Yes, I have the same issue. I tested it in docker. 

```
root@stmoon-XPS-13-9370:~/multi_vehicle/scenario1# netstat -antup | grep 1857
udp        0      0 0.0.0.0:18570           0.0.0.0:*                           11771/../bin/px4    
udp        0      0 0.0.0.0:18571           0.0.0.0:*                           11958/../bin/px4    
udp        0      0 0.0.0.0:18572           0.0.0.0:*                           32622/../bin/px4    
udp        0      0 0.0.0.0:18573           0.0.0.0:*                           12187/../bin/px4    
udp        0      0 0.0.0.0:18574           0.0.0.0:*                           12343/../bin/px4        
root@stmoon-XPS-13-9370:~/multi_vehicle/scenario1# netstat -antup | grep 1428
udp        0      0 0.0.0.0:14280           0.0.0.0:*                           11771/../bin/px4    
udp        0      0 0.0.0.0:14281           0.0.0.0:*                           11958/../bin/px4    
udp        0      0 0.0.0.0:14283           0.0.0.0:*                           12187/../bin/px4    
udp        0      0 0.0.0.0:14284           0.0.0.0:*                           12343/../bin/px4    
root@stmoon-XPS-13-9370:~/multi_vehicle/scenario1# netstat -antup | grep 1458
udp        0      0 0.0.0.0:14580           0.0.0.0:*                           11771/../bin/px4    
udp        0      0 0.0.0.0:14581           0.0.0.0:*                           11958/../bin/px4    
udp        0      0 0.0.0.0:14583           0.0.0.0:*                           12187/../bin/px4    
udp        0      0 0.0.0.0:14584           0.0.0.0:*                           12343/../bin/px4    
root@stmoon-XPS-13-9370:~/multi_vehicle/scenario1# 
```

Jaeyoung-Lim, 2020-09-06:
@stmoon I was wondering if the latest changes you did is triggering this, since you are the last person modifying the script

Jaeyoung-Lim, 2020-09-06:
@stmoon @lappie @KrisRadowski Unfortunately, I cannot reproduce this issue. This might be something coming from the system.

I am running it on Ubuntu 20.04, Gazebo 11.1.0

lappie, 2020-09-14:
With a fresh install of gazebo 11 it now works.

closed 2020-09-29 by Jaeyoung-Lim