PX4/PX4-Autopilot #15612: Possible attitude estimation bug

petrlmat, 2020-08-25:
**Describe the bug**
After ~30s of full manual indoor flight (~5m height above floor), the estimated attitude began to drift significantly in the roll angle (even in pitch, but not so much), even though the pose desired by the pilot was the hover state. The pilot tried to compensate by tilting the drone in the opposite direction, however, he reached the limit position of the roll stick on the RC and the roll still continued to drift. The high velocity led to a rough crash into a wooden bench. There was no RC signal loss and we checked that the RC had no trims and subtrims in effect.

This was the first time we experienced this behavior after ~4 years working with PX4 firmware.

The second incident we had was the next day on the same location with different hardware, but the same symptoms. After ~30s of manual flight in altitude mode (~5m height above floor) the UAV started tilting uncontrollably and then crashed into a wall. Again no RC signal losses, no trims and subtrims. We were even flying with this platform the same day in the morning for ~10 minutes without any issues.

Please find attached logs and description of the hardware in the following sections. My colleague will also upload a video which compares the PX4 estimated attitude with the correctly estimated attitude obtained from a custom attitude estimator run on the recorded IMU flight data.

What is the cause of this behavior? Could there be a bug in the attitude estimator that only manifests in some highly specific conditions? Is there some problem apparent from the logs that we are missing?

Please ask for any relevant additional information that could help solving the issue. Until we resolve this issue we cannot fly indoor in this setup, because it is too unreliable and dangerous.

**To Reproduce**
We tried to reproduce the behavior in lower height above floor with another drone with the same setup as in the first flight without success. 

**Log Files and Screenshots**
[Flight 1](https://review.px4.io/plot_app?log=70d80c6c-46f4-4d45-a710-bcc0ed0cae1c)
[Flight 2](https://review.px4.io/plot_app?log=5dc3d4bb-5c16-4ec6-b7ad-17f2456e4f3b)

**Drone (please complete the following information):**

***Flight 1***
- Frame: Quadrotor Tarot T650
- Total mass: 3.5 kg
- Motors: Tarot 4114-11 320KV
- ESCs: BL-Heli 32 50A
- FCU: Pixhawk 4 (PX4 FW v1.10.1)
- RC: Hi-Tec Flash 8
- RC receiver: Optima SL

***Flight 2***
- Frame: Custom coaxial octarotor
- Total mass: 5.5 kg
- Motors: T-Motor MT2814-11 710KV
- ESCs: FOXY Multi 30A Opto V2015
- FCU: Pixhawk 2.1 Cube (PX4 FW v1.10.1)
- RC: Hi-Tec Aurora 9
- RC receiver: Optima SL

**Additional context**
- Magnetometer fusion was turned off for both flights
- All components were used previously in many flights
- Both FCUs are attached to the frame by screws to keep them fixed
- Calibration of accelerometers, gyroscopes and level horizone was done the day before the first incident

dagar, 2020-08-25:
Here's the ecl analsys script output from Flight 1.
[70d80c6c-46f4-4d45-a710-bcc0ed0cae1c.ulg.pdf](https://github.com/PX4/Firmware/files/5124672/70d80c6c-46f4-4d45-a710-bcc0ed0cae1c.ulg.pdf)

It looks like something went wrong with the IMU bias estimation.

[image]

Similarly, here's Flight 2.
[5dc3d4bb-5c16-4ec6-b7ad-17f2456e4f3b.ulg.pdf](https://github.com/PX4/Firmware/files/5124687/5dc3d4bb-5c16-4ec6-b7ad-17f2456e4f3b.ulg.pdf)

[image]
[image]

Finally, here's what it looked like in the OK flight.
[e0c02122-4335-4ce3-ab3d-598a38e98c7f.ulg.pdf](https://github.com/PX4/Firmware/files/5124709/e0c02122-4335-4ce3-ab3d-598a38e98c7f.ulg.pdf)
[image]

dagar, 2020-08-25:
@petrlmat I'd still like to try and understand the root cause of your issue with v1.10.1, but in the meantime if you're able to test relatively easily it would be extremely helpful if you could try v1.11.0-rc3 (about to become v1.11.0 stable) to see if it's still a problem. There's been quite a lot of improvement in the estimator and sensor pipeline since v1.10.

petrlmat, 2020-08-26:
Thank you for the quick answer! As I said, we tried to reproduce the error later without success. We want to do more flights in similar setup in about a month, so we will try with 1.11.0. However, we need to carry some expensive equipment, which is too risky until we debug the issue.

closed 2023-02-07 by bresch