PX4/PX4-Autopilot #15619: HP: requesting home position, EKF2 not working when publishing position via Vicon MOCAP

Bionic-Sock, 2020-08-25:
I have tried several things for the past few weeks and cant seem to get EKF2 working on my quadrotor. I had some success while publishing the '/mavros/fake_gps/mocap/pose', but had problems getting current data to hardware. The data received was really delayed, so the quad would go into position HOLD, but obviously didn't have the ability of staying in one position and would veer off in a random direction. I struggled with this delay for a few days before trying to go a different route and publish to  '/mavros/vision_pose/pose'. By using the MAVlink inspector in QGroundcontrol, I notice that Odometry is getting correct values around 30Hz, but Mavros keeps displaying "[ **INFO] [1598390189.698583264]: HP: requesting home position**" and wont go into position hold. I've also tried working with 'mavros/odometry/out' but also had no luck.

[px4_config_yaml.txt](https://github.com/PX4/Firmware/files/5126530/px4_config_yaml.txt)
[mavros_output.txt](https://github.com/PX4/Firmware/files/5126431/mavros_output.txt)
[qgroundcontrol_params.txt](https://github.com/PX4/Firmware/files/5126466/qgroundcontrol_params.txt)
[vision_pose_topic_output.txt](https://github.com/PX4/Firmware/files/5126481/vision_pose_topic_output.txt)

I am at a loss and I don't know where to go next. I am willing to try fake_gps if thats an easier fix or I'm willing to stick with what I'm currently doing with trying to publish to '/mavros/vision_pose/pose' or 'mavros/odometry/out'.

Thanks.

Bionic-Sock, 2020-08-25:
I also followed this issue [https://github.com/PX4/Firmware/issues/13510](url) and had no luck.

dagar, 2020-08-28:
I believe this is more of a Mavros problem, but I could be wrong. The vehicle isn't going to have a home position set unless there's a valid global position estimate, which you're not going to have in this scenario.

Bionic-Sock, 2020-09-02:
So I updated the method I was using. Now I switched back to fake_gps using a vicon motion capture software. I am finally able to get a valid position lock on the quadrotor and change into hold, BUT when flying in position hold mode the quadrotor drifts very fast in one direction leading to a crash. The quadrotor flies fine in manual mode, but is incapable of flying stable in hold mode. Anyone know what to do?

frekabi, 2021-05-16:
I have the same problem, I need to use EKF2 vision fusion and I have searched a lot for a solution. It seems that the fake_gps is going to deprecate and it has lots of problems. If anyone has a successful plan for implementing vision based positioning with px4 EKF2 please help!!!!!

DavidePatria, 2022-02-04:
@frekabi have you been able to solve this?

frekabi, 2022-02-04:
Hi David
unfortunately, I couldn't solve that issue, I tried everything I found in
forums but nothing worked for me even the fake GPS. Please let me know if
you find any solution.
Regards
Fatemeh

Isaac-Seslar, 2022-02-05:
Hello @frekabi and @DavidePatria,

It's been quite some time since I've worked on this, but I did find the solution. You'll have to forgive me because I don't have the code in front of me.

You basically need to edit your px4_pluginlist.yaml (or apm_pluginlist.yaml if you're using ardupilot) file and blacklist all of the things you aren't using. My problem was something was whitelisted that required a home position and wouldn't let anything else run.

Sorry I can't be more help, but it at least this will give you a place to start. 

Let me know how it goes.

Isaac-Seslar, 2022-02-05:
Also, I wouldn't use fake_gps. I think I used the vision_pose/pose topic if I remember correctly

DavidePatria, 2022-02-05:
@Isaac-Seslar thank you for your tips. I'm a bit confused though, as the different options to use external position sources, such as mocap and vision, are available in the config file as suboptions for fake_gps, so it seems that these options belong to that plugins and are not stand alone. I will research a bit to find a list of existing plugins, to try and follow your advice.

vipin3690, 2022-10-10:
Are you able to solve the issue

DavidePatria, 2022-10-11:
@vipin3690  you should look how it is done [here](https://github.com/mzahana/px4_fast_planner), it doesn't use mocap topic

bresch, 2023-04-25:
Have you tried setting the global origin using https://mavlink.io/en/messages/common.html#SET_GPS_GLOBAL_ORIGIN in v1.14 (not supported in older versions)?

CSU-SS, 2023-06-21:
@Isaac-Seslar Hello, have you finally solved the problem? now, I am facing the same question.