PX4/PX4-Autopilot #15625: EKF2 doesn't fuse rangefinder measurements

ilNonn0, 2020-08-26:
I am facing a problem trying to integrate the rangefinder measurement into the EKF2 estimate.
Following the documentation, I set up the system to use the LL40LS as primary source of information for the hight estimation, but the EKF2 still use the barometric sensor.
Then I tried all the possible parameter combinations, but still, EKF2 use the barometer.
Here what I got in a test flight: 
_INFO  [ecl/EKF] 9801942: EKF aligned, (baro height, IMU buf: 22, OBS buf: 14)_

Is there a bug that prevents EKF2 to use the rangefinder to estimate the drone altitude?

here the log file for a more detailed picture: https://review.px4.io/plot_app?log=216a2be0-6655-4a49-a462-f7b03d68e036

Thanks in advance.

ilNonn0, 2020-08-27:
Solved
The sensor was wrongly initialized (in horizontal position)

ilNonn0, 2020-08-27:
in particular, it works as a distance sensor to the ground, initializing with the following command:
ll40ls start -X -b 5 -R 25
The default value ( equal to 0) doesn't work in my case

bresch, 2020-08-31:
fixed by https://github.com/PX4/Firmware/pull/15639

closed 2020-08-31 by bresch