PX4/PX4-Autopilot #15627: no Position data from ODOMETRY.

farhangnaderi, 2020-08-26:
**ODOMETRY data cannot be fused on Pixhawk on the latest Master but it does on the Stable Version.**

**To Reproduce**
 Following the procedures to get Position data from Realsense camera and by starting MAVROS and Auterion VIO bridge. 

**Expected behavior**
X and Y are zero in Mavlink Inspector when on master Firmware.

**Log Files and Screenshots**

With the Stable Release,

https://logs.px4.io/plot_app?log=20166f3d-a58f-4c5d-83b3-3476ce3ea6fc

[image]

With the Master:

https://logs.px4.io/plot_app?log=1566bab0-6931-49f3-8c2c-8ad775b57039

[image]

**Drone**

F450 Frame

Tested on both PixRacer and Cube Black

kamilritz, 2020-08-26:
The local position x and y will always be set to zero, when there is no position or velocity information fused.

From the log, I see that you set the EKF2_AID_MASK to 344, meaning you enabled `EV_YAW` and `ROTATE_EV` at the same time. Checking both options (bit 4 and 6) is not supported, neither is it tested. Please follow the [documentation](https://hamishwillee.gitbooks.io/ham_px4userguide/content/en/advanced_config/tuning_the_ecl_ekf.html#ekf2_extvis) on how to setup the EKF2_AID_MASK and try again.

farhangnaderi, 2020-08-26:
Indeed, while it's been tested with both 280 and 28 as well.
On the other hand, one release is doing it with the same params.

farhangnaderi, 2020-09-01:
@kamilritz The problem is still there with the AID masks you mentioned. But at the same time everything ok with 1.10 stable release.

kamilritz, 2020-09-05:
@farhangnaderi is the behavior still exactly the same? when not, do you have a log with the correct AID mask?

farhangnaderi, 2020-09-11:
@kamilritz 
Below are with 4 different possibilities:

AID MASK 24:

https://logs.px4.io/plot_app?log=c5798d8d-53e3-4ac3-bce2-d170e0bab6ae

AID MASK 72:

https://logs.px4.io/plot_app?log=be0fe116-a331-421a-b019-0051c0ca0c61

AID MASK 272:

https://logs.px4.io/plot_app?log=f266993d-ccf6-4462-acfd-97815c430eef

AID MASK 280:

https://logs.px4.io/plot_app?log=27572b38-7450-4cbc-8ce1-8651076562f7

Please let me know if I am wrong somewhere.

potaito, 2020-11-17:
@farhangnaderi 
> With the Stable Release,
> 
> https://logs.px4.io/plot_app?log=20166f3d-a58f-4c5d-83b3-3476ce3ea6fc
> 
> 
> With the Master:
> 
> https://logs.px4.io/plot_app?log=1566bab0-6931-49f3-8c2c-8ad775b57039
> 

In the second log, the one from Master, I don't even see the `vehicle_visual_odometry_*` messages at all. Is that part of the problem?

Phillweston, 2021-01-24:
The VIO code, the author reverted the merge, and the latest VIO code still cannot get the odometry information.

Phillweston, 2021-01-24:
> The VIO code, the author reverted the merge, and the latest VIO code still cannot get the odometry information.

Is there a way to achieve the function of VIO odometry without downgrading the firmware to 1.10?

I found the reason as follows:
https://github.com/Auterion/VIO/pull/15#issuecomment-718905186

farhangnaderi, 2021-01-24:
I will try with different Firmware Versions and if ok, I will close this issue.

farhangnaderi, 2021-01-27:
I will open another issue for the new Firmware 1.11.3.

closed 2021-01-27 by farhangnaderi

adamdai, 2021-05-12:
Is there an update on this? I am using 1.11.3 and still running into the same issue (using Optitrack-Mocap pose estimate). Is there a link to the new issue?

farhangnaderi, 2021-05-12:
@adamdai plz check with 1.12 as well. I guess it was fixed.

adamdai, 2021-05-12:
@farhangnaderi thanks for the quick reply. we just updated to 1.12.0 but are still not seeing the odometry messages. We have set the MAV_ODOM_LP parameter to 1, and are publishing to the /mavros/vision_pose/pose topic.

Peter010103, 2021-09-01:
@adamdai Hey, we're trying to use the Opitrack system as well and are sending the pose estimates using the mavros_extras `vision_pose_estimate` plugin. Is this what you were doing too?

Once we start our mavros node, we can see the `LOCAL_POSITION_NED` Mavlink message updating at 30Hz from the Mavlink Inspector in QGC, but we are also not seeing the `ODOMETRY` Mavlink message even with the `MAV_ODOM_LP = 1`. My question is a) does the drone need to receive this Mavlink topic and b) were you able to get your system flying using Position/ Offboard mode without `ODOMETRY?`

Peter010103, 2021-09-02:
@fabrizioromanelli Thanks for the reply.

From the [external_position_estimation](https://docs.px4.io/master/en/ros/external_position_estimation.html) page on the docs, it says MAVLink can except either `VISION_POSITION_ESTIMATE` or `ODOMETRY`, which are mapped to the `vehicle_visual_odometry` uORB message. Looking at the [Mavlink messages page](https://mavlink.io/en/messages/common.html#VISION_POSITION_ESTIMATE), the only difference between the two messages seem to be that `ODOMETRY` includes velocity information as well as position.

So are you suggesting I need to send both Mavlink messages (via mavros or otherwise). How did you solve this problem without `ODOMETRY`? Thanks :)

fabrizioromanelli, 2021-09-03:
@Peter010103 in our implementation, we did not use Mavlink messages, but we went for uORB messages directly but only sending the `VISION_POSITION_ESTIMATE` and not the `ODOMETRY` one. I know that Mavlink uses a slightly different approach but as I didn't use it, I cannot say how you can operate with that.

pitergit95, 2022-04-11:
Hello, 

Unfortunately your solution doesn't work for me. I have tried:
Both Firmwares - 1.10.1 and 1.12.3
EKF2_AID_MASK - 345 and 73
Both topics - /mavros/odometry/out and /mavros/vision_pose/pose
Hardware - Cube Orange, 3DR Pixhawk 1, Pixhawk 4

I have used various combinations and it didn't work for me. Could you please tell me which configuration should work, or tell me which configuration works for you? 

It would be very helpful, thanks

mamariomiamo, 2022-04-13:
Running v1.12.3, odometry is not showing up in qgroubndcontrol Mavlink inspector even though MAV_ODOM_LP is set to 1.
In Mavlink Console, uorb top shows there is vehicle_visual_odometry publishing

pitergit95, 2022-04-14:
Sorry but i'm not familiar with uORB messaging. To be sure, does it mean that VIO works correctly with Pixhawk, without being visible in Mavlink inspector?