PX4/PX4-Autopilot #15628: Broken manual and position control of Rover

SalimTerryLi, 2020-08-27:
**Describe the bug**
Commit 53b3dfa64ea85876014c5743004b47173bb88007 makes rover lost the ability of manual control, as well as position setpoint.

**To Reproduce**
1. Checkout to that commit by `git checkout 53b3dfa64ea85876014c5743004b47173bb88007`
2. Execute `make px4_sitl gazebo_rover`
3. Open QGC and enable virtual sticks.
4. Arm the vehicle in manual mode, then move around sticks. Nothing happened.
5. Switch to Hold mode. Click on the map and slide the "GO TO" bar. Nothing happened either.

Checkout to the previous commit. Everything works well.

@dagar

dagar, 2020-08-27:
Thanks for reporting, I'll take a look.

dagar, 2020-09-02:
https://github.com/PX4/Firmware/pull/15664

closed 2020-09-02 by dagar