PX4/PX4-Autopilot #15629: PX4 crash due to I2C malfunction

VinniF, 2020-08-27:
Hi all,

I am reasearcher at the University of Stuttgart in Germany. We use the PX4 system on our Pixhawk drone development where we write our own modules but use the general functionality of PX4 and nuttx.

Recently, we crashed an aircraft due to the Pixhawk FMU chip crashing. We have probably found the reason for this crash and we believe that this could be solved softwarewise.

Error: Pixhawk FMU module crashes when the throttle is raised too high, error appears very sporadically and is difficult to reproduce.

Description: A magnetic interference between a high power cable and the I2C bus makes the Pixhawk FMU crash. 

Repro steps:
Minimal working example: Power a Pixhawk with a dedicated LiPo and PM07. Only connect Holybro Digital Air Speed Sensor to the Pixhawk and conect via I2C. Use a cable that is slightly too long and create one loop with the I2C cable. Connect an RC engine to a seperate set of batteries (we use xpxs) and run these power cables over the I2C coil. Run the engine with about 30-70 A of current. If the error does not happen, you might need to adjust the I2C cable and power cables a little bit. They need to be very close to each other.
 
Actual Result: Pixhawk continues to operate nominally
Expected Result: Pixhawk FMU shuts off

Why do I think this can be solved: The I2C bus is quite sensitive to disturbances. The incoming signal through I2C should be checked for signal errors, noise, disturbances. If any unusual behavious is found, the I2C should be reset. In the worst case, the I2C should be turned off and the sensor disregarded. It must not happen that the I2C kills the entire FMU.
 
Environment
We branched our PX4 firmware from d36c0e131dd7a1df78cad313c07d6ae8bdeaf950
Pixhawk 4 FMUv5 
Anything else?

PX4 log: https://logs.px4.io/plot_app?log=54757cef-3b2c-4803-9cbe-737a17e9b37d

dagar, 2020-08-30:
What does the actual crash look like? Do you have a hardfault log anywhere?

LorenzMeier, 2020-08-30:
Can you confirm that your tree has this commit in? Please share the commit hash of the NuttX subdirectory.
https://github.com/PX4/NuttX/commit/1f2f6d8cbe7c85f30fd75ec53c20566e91b14f21

VinniF, 2020-08-31:
> What does the actual crash look like? Do you have a hardfault log anywhere?

I have never recorded a hardfault log before. Besides the flight log, we have no log available. Because of the behaviour of the Pixhawk after that crash, we are sure that the FMU quits its job and the IO controller takes over. In order to reproduce the error reliably and test workarounds, we reset the FMU with the external FMU reset and the behaviour is the same.

> Can you confirm that your tree has this commit in? Please share the commit hash of the NuttX subdirectory.
> [PX4/NuttX@1f2f6d8](https://github.com/PX4/NuttX/commit/1f2f6d8cbe7c85f30fd75ec53c20566e91b14f21)

This commit is included in our build - we are just one commit younger than that: [PX4/NuttX@ea24b54](https://github.com/PX4/NuttX/commit/ea24b543df6aab6607e1fa758e4a7586b9ed3113)

dagar, 2020-08-31:
> I have never recorded a hardfault log before. Besides the flight log, we have no log available. Because of the behaviour of the Pixhawk after that crash, we are sure that the FMU quits its job and [...]

Is that the actual flight log where the FMU stopped responding? Can you check the actual sdcard for other files (hardfault_log)?

VinniF, 2020-09-01:
> Is that the actual flight log where the FMU stopped responding? Can you check the actual sdcard for other files (hardfault_log)?

Yes, I will check th

VinniF, 2020-09-02:
Hi dagar,

I just checked the SD card. Yes, the flight log linked above is the actual flight log where the FMU stopped responding. I checked the SD card and there are a few hardfault logs but none of them with the correct date - I believe that no hardfault logs were produced. The existing hardfault logs are from an earlier version and test where the stack was not sufficient.

dagar, 2020-09-08:
I'm sorry, but I don't think there's enough to go on from that flight log alone. Do you think you could reproduce it on the bench? If so we could connect the serial console and possibly a debugger to figure out the root cause.

davids5, 2020-09-08:
FYI: For the upgrade, I tested the F7 I2C driver again, It can not get it to lock up adding random pulses on SDA, If you can capture the bus with a scope/logic analyzer  on the failure case, or provide a simple way to trigger the issue that would help.

LorenzMeier, 2021-01-01:
@VinniF Can this still be reproduced?

VinniF, 2021-01-12:
@LorenzMeier
Hi Lorenz, yes this bug can still be reproduced under the circumstances I mentioned above. But we have not updated our firmware, we are still on the same version.

mrpollo, 2021-02-03:
We stress-tested with an I2C noise injector, couldn't reproduce and we haven't seen any more reports, we are closing the issue.

If you can share pictures and help us reproduce we can re-open and work with you to get this fixed.

closed 2021-02-03 by mrpollo