PX4/PX4-Autopilot #15645: Define VTOL takeoff alt and next waypoint alt

moreba1, 2020-08-29:
Hi

There is an issue with the current takeoff and transition planning for VTOLs in px4. takeoff alt and next waypoint alt are same, during transition altitude will increase but when transition completed altitude must be decreased to reach the previous alt.
it is dangerous and is not good behavior. we must define the alt of the next waypoint independently and a little more.

[image]

bresch, 2020-08-31:
I think this is specific to your setup having too much lift during the front transition. Can you confirm that this isn't the case?

moreba1, 2020-08-31:
> 
> 
> I think this is specific to your setup having too much lift during the front transition. Can you confirm that this isn't the case?

This problem may occur for any quad plane VTOL drone.

saengphet, 2020-09-01:
> Hi
> 
> There is an issue with the current takeoff and transition planning for VTOLs in px4. takeoff alt and next waypoint alt have the same value. during the transition, altitude will increase but when transition completed altitude must be decreased to reach the setpoint alt.
> it is dangerous and is not good behavior. we must be able to define the altitude of the next waypoint independently and a little more.
> 
> [image]

We are facing this behavior during the flight test. Is it possible to let the UAV continue to the next waypoint without altitude correction "if current altitude > transition altitude" after the transition is completed, if not try to increase altitude before going to the next waypoint?

sfuhrer, 2020-09-01:
@moreba1 There really shouldn't be that much altitude gain during the transition. If you could provide a log file I can check for the reason of the climb, and maybe help with the tuning. 
Nevertheless, I agree that we could change things on the PX4 side (in combination with QGC if required) to improve the UX. E.g. we could neglect the location specified with the VTOL takeoff waypoint, and instead make it accepted once the transition is completed. The vehicle will then directly fly to the first actual waypoint instead of having to fly to the VTOL transition waypoint in fixed-wing first. 
Other ideas?

moreba1, 2020-09-01:
>  I agree that we could change things on the PX4 side

Thank you. changing code is better than tuning because maybe every platform needs especial tuning for this issue.