PX4/PX4-Autopilot #15649: Estimation drifting with GPS & Vision Velocity

nbelanger99, 2020-08-31:
Using v1.10.2, built for fmu-v5 on generic quad.

We're doing some testing with visual velocity feedback + RTK GPS. What we've found is counterintuitive; the estimation is better with just RTK (EKF2_AID_MASK=1 decimal) than with RTK & VIO (EKF2_AID_MASK=321 dec). With the latter configuration, it seems to drift drastically, which leads me to suspect that RTK is being ignored for some reason when fusing visual velocity.

Another oddity is that I can't find the bit of code which specifically handles this logic. I know that visual position data fusion logic happens in control.cpp, but even when I search for the string "commencing external vision velocity fusion" (from PX4 flight log), I can't find it in the file, or even anywhere in the codebase, and the fusion_mode bitmask doesn't seem to account for it. 

If anyone's familiar with this logic, and could help me find out why EKF2 isn't apparently utilizing RTK to account for drift, that would be quite helpful.

AlexisTM, 2020-09-02:
Maybe you are looking for ecl, the estimation library used for estimation?

dagar, 2020-09-08:
You'd need to share a representative flight log for someone to take a closer look. https://logs.px4.io/

I would also suggest you try again with current PX4 (v1.11.0) or master. There's been quite a lot of estimator work since the v1.10 release.