PX4/PX4-Autopilot #15667: USB connection is only accessible if connected in a window of few seconds

AlexisTM, 2020-09-02:
**Describe the bug**

If we do not connect the USB to a consumer right after the bootloader, once the RGB LED starts breathing blue (in a window of few seconds), the Pixhawk is inaccessible, no message can be received or sent anymore, but the TTY ACM driver is still up and running, thinking it can communicate.

I could replicate the exact same behavior going one level down, using `libusb1.0` instead of using the kernel `usb-cdc-acm` driver but the exact same behavior happened. This seems to indicate that the last burst would come from the Pixhawk's buffer and not the tty driver buffer. As the log shows below, if nobody subscribes to the data in `/dev/ttyACM0`, the USB connection is not initialized.

**To Reproduce**
Steps to reproduce the behavior:
1. Power the Pixhawk with the USB
2. Wait for the LED to be blue breathing
3. Wait an additional 10 seconds
4. Try to connect Mavros, QGc or `cat /dev/ttyACM0`
5. When connecting, few messages are received
6. The connection hangs: No message can be sent or received

Steps to reproduce the behavior after successful connection:
1. Power the Pixhawk with the USB
2. Wait for the LED to be blue breathing
3. ~Wait an additional 10 seconds~
4. Try to connect Mavros, QGc or `cat /dev/ttyACM0`
5. The connection is successful and can be used for any time you want
6. Disconnect (stop the node)
7. Wait 10 seconds
8. Try to reconnect with the same method
9. When connecting, few messages are received
10. The connection hangs: No message can be sent or received

**Expected behavior**

Steps to reproduce the behavior after successful connection:
1. Power the Pixhawk with the USB
2. Wait for the LED to be blue breathing
3. You can wait any time you want before connecting
4. Try to connect Mavros, QGc or `cat /dev/ttyACM0`
5. The connection is successful and can be used for any time you want

**Log Files and Screenshots**

`sudo apt install vusb-analyzer usbmon`

Following a USB log file showing the problem. The Pixhawk in this log is DEV 1018. To view it, download it, rename it to remove the `.txt` and open it with vusb-analyzer:

[usbmon.mon.txt](https://github.com/PX4/Firmware/files/5160810/usbmon.mon.txt)

```bash
mv usbmon.mon.txt usbmon.mon # .mon files are not supported on Github
vusb-analyzer usbmon.mon
```

*History of the log:*
* `0.00s`: Start the log
* `6.00s`: Connect Pixhawk
* `12.50s`: Pixhawk left the Bootloader (CDC driver advertised)
* `13.50s`: Mavlink node started on `/dev/ttyACM0` (showing normal behavior)
* `21.25s`: Mavlink node manually stopped
* `37.70s`: Mavlink node started on `/dev/ttyACM0` (showing a stuck behavior)
    * The messages **do contain data**

Normal start:
[image]

Failing start:
[image]

**To reproduce the log:**

```bash
sudo su
modprobe usbmon
cat /sys/kernel/debug/usb/usbmon/1u > usbmon.mon
```

**Drone:**
- Pixhawk 1 (original 3DR)
- Firmware: px4_fmuv2_default

dagar, 2020-09-08:
Nothing immediately comes to mind here, but I'd suggest you try to instrument the mavlink module and debug through the serial console (or debugger) to see the USB mavlink instance is doing.

dagar, 2020-09-08:
> **To Reproduce**
> Steps to reproduce the behavior:
>
> 1. Power the Pixhawk with the USB
> 2. Wait for the LED to be blue breathing
> 3. Wait an additional 10 seconds
> 4. Try to connect Mavros, QGc or `cat [...]

In step 5 do you happen to know specifically which messages are getting through?

AlexisTM, 2020-09-08:
The messages are the typical messages the Pixhawk was supposed to send such as IMU data, position data, ... 

Note that I do not have a serial cable as of now and cannot debug further. :disappointed: 

Could it be caused by the age of the Pixhawk? It is an old 3DR Pixhawk.

dagar, 2020-09-14:
> Could it be caused by the age of the Pixhawk? It is an old 3DR Pixhawk.

Unlikely unless the physical connection is questionable.

AlexisTM, 2020-09-15:
Using the same cables with other USB-CDC devices, I do not have the issue. (tested with 3 other devices using CDC)

dagar, 2020-11-12:
Possible solution - https://github.com/PX4/PX4-Autopilot/pull/16180.

AlexisTM, 2020-11-13:
@dagar I will test this! Thanks!

AlexisTM, 2020-11-16:
[Closing this in the meantime]

closed 2020-11-16 by AlexisTM