PX4/PX4-Autopilot #15683: generalized per sensor instance configurable metadata

dagar, 2020-09-04:
- IMU
    - EKF2_IMU_POS_{X,Y,Z}
        - **TODO:** Move to per instance `SENS_IMUx_POS{X,Y,Z}`? 
  - magnetometer
    - external rotation, priority
  - barometer
    - priority, maybe an offset
    - **TODO:** Create barometer calibration (`CAL_BAROx`) and populate somehow
  - range finders
    - orientation
        - **TODO:** Move to per instance `SENS_RNGx_ROT`? 
    - EKF2_RNG_POS_{X,Y,Z}
        - **TODO:** Move to per instance `SENS_RNGx_POS{X,Y,Z}`? 
  - optical flow
    - SENS_FLOW_ROT
        - **TODO:** Move to per instance `SENS_FLOWx_ROT`? 
    - EKF2_OF_POS_{X,Y,Z}
        - **TODO:** Move to per instance `SENS_FLOWx_POS{X,Y,Z}`? 
  - differential pressure
    - per instance calibration
         - what about model (CAL_AIR_CMODEL), tube length (CAL_AIR_TUBELEN), tube diameter (CAL_AIR_TUBED_MM)?
 - GPS
    - device ids needed
         - **TODO:** extend device ids to GPS and UARTS as a bus option?
    - EKF2_GPS_POS_{X, Y, Z}
        - **TODO:** Move to per instance `SENS_GPSx_POS{X,Y,Z}`? 

For all sensor types without an explicit calibration step how/when do we have the system scan everything that current exists and populate the parameters?

Another idea is that each sensor instance could have a configurable optional/required parameter that would roll into preflight checks.

dagar, 2020-09-04:
FYI @priseborough @CarlOlsson @bkueng @bresch

JacobCrabill, 2020-11-25:
I think it could be done similarly to calibration, in that if an SENSx_ID parameter is 0, it will take the ID of the next unused uORB instance of the sensor it is for.  Could this be done in the sensors module as each new uORB instance is discovered?  Perhaps it could send a MAVLink message to the GCS if user configuration is needed as new sensor instances are "discovered" - i.e. a new rangefinder comes online on the UAVCAN bus and takes the RNG1 parameter instances, it defaults to downward-facing at position (0, 0, 0), and the user is notified that the RNG1_ROT and RNG1_{x,y,z} parameters exist and can be set if needed.

On another note, are there any thoughts on allowing a semi-dynamic creation of these parameters?  e.g. instead of spelling out IMU0, IMU1, IMU2, IMU3 parameters, define a general IMU{n} parameter that allows up to some number of instances at runtime.  Not sure how this would actually work, but the existing approach just feels... overly verbose.