PX4/PX4-Autopilot #15686: In HIL mode, motor will be disarmed after 10s even if disable the auto disarmed function on the QGC

kcx064, 2020-09-05:
In HIL mode, motor will be disarmed after 10s even if I disable the auto disarmed function on the QGC

kcx064, 2020-09-05:
px4 version is 1.11.0

dagar, 2020-09-05:
What are you using for HIL? Can you share the log with this happening? https://logs.px4.io/

kcx064, 2020-09-06:
Thank you for your reply. I used Airsim v1.3.1 for HIL.  The log is [https://logs.px4.io/plot_app?log=28740df9-a979-40b0-82aa-bf7cf7cd6eb3](https://logs.px4.io/plot_app?log=28740df9-a979-40b0-82aa-bf7cf7cd6eb3)

This issue maybe start with v1.11beta2

kcx064, 2020-09-06:
My hardware is **Cube Black** and **Durandal**. They are both have the issue.

Jaeyoung-Lim, 2020-09-06:
@kcx064 Can you try and build Airsim from source? I think there was a issue with the v1.3.1 release of Airsim and was fixed afterwards

kcx064, 2020-09-06:
> @kcx064 Can you try and build Airsim from source? I think this is most likely a Airsim issue since there was some issue with the v1.3.1 release of Airsim and was fixed afterwards

Thank you! But PX4 v1.10.1 does not have the issue when I used Airsim v1.3.1

Jaeyoung-Lim, 2020-09-06:
@kcx064 Yes, and that is why something had to be fixed on the Airsim side

kcx064, 2020-09-06:
> @kcx064 Yes, and that is why something had to be fixed on the Airsim side

It is to say PX4 v1.11 makes some modification for HITL mode which makes Airsim can not work properly in HITL mode, so Airsim need to be fixed?

dagar, 2020-09-06:
> Thank you for your reply. I used Airsim v1.3.1 for HIL. The log is https://logs.px4.io/plot_app?log=28740df9-a979-40b0-82aa-bf7cf7cd6eb3
> 
> This issue maybe start with v1.11beta2

"Disarmed by lockdown"

[image]

You're hitting this. https://github.com/PX4/Firmware/blob/af84f449c74b1b108bf23f6b8fdf63cb95c1b39a/src/modules/commander/Commander.cpp#L1681-L1692

In HIL lockdown is enabled for safety. Let me see about a quick fix.

@Jaeyoung-Lim we should talk about getting automated HIL tests in CI. One helpful step in that direction would be if we make it as trivial to launch HITL (gazebo or jmavsim) as SITL gazebo.

Jaeyoung-Lim, 2020-09-06:
> @Jaeyoung-Lim we should talk about getting automated HIL tests in CI. One helpful step in that direction would be if we make it as trivial to launch HITL (gazebo or jmavsim) as SITL gazebo.

Okay, makes sense. Do we want to hook this up to github actions?

dagar, 2020-09-06:
The lockdown -> auto disarm was introduced in https://github.com/PX4/Firmware/pull/14766.

TSC21, 2020-09-06:
> @Jaeyoung-Lim we should talk about getting automated HIL tests in CI. One helpful step in that direction would be if we make it as trivial to launch HITL (gazebo or jmavsim) as SITL gazebo.

@dagar @Jaeyoung-Lim,
@bperseghetti is already working on those. And he is facing the exact same problem as above. @bperseghetti do you mind syncing with @dagar on a fix for this? Thanks!

dagar, 2020-09-06:
> > @Jaeyoung-Lim we should talk about getting automated HIL tests in CI. One helpful step in that direction would be if we make it as trivial to launch HITL (gazebo or jmavsim) as SITL gazebo.
> 
> Okay, makes sense. Do we want to hook this up to github actions?

Let's get the pieces in place to do it as a developer first, then figure out how to make it live in github actions or Jenkins (attaching to hardware is a "detail").

1. Is sitl_gazebo HIL good enough to start using regularly? How can we make it trivial to launch directly and specify the serial port? What about included airframes that correspond to sitl_gazebo models? I don't think they're in the NuttX ROMFS.
2. Scripting the process.
    - flashing, setting the right airfarme, connecting
    - launch mission/mavsdk test and monitor
    - transfer ulg over mavlink

We can keep discussing outside of this issue.

dagar, 2020-09-06:
@kcx064 can you try https://github.com/PX4/Firmware/pull/15690?

kcx064, 2020-09-06:
> @kcx064 can you try #15690?

It's my pleasure!

kcx064, 2020-09-06:
> @kcx064 can you try #15690?

Thank you!  I have try your branch,  it work properly in hil mode!

bperseghetti, 2020-09-06:
@dagar @TSC21 @Jaeyoung-Lim I had noticed this issue with HITL using gazebo last Monday. I think there is a lot of ways we can automate HITL testing. I had "ugly fixed" it by just commenting out the autodisarm and was looking through the rest of the HITL pipeline to see more about avoiding edge cases and how well it can "represent" the actual system before issuing an actual fix for that specific bug. I pulled and tested the bug for HITL/gazebo and it works as well.

commit 1e421711 (2020-09-06): commander: HITL skip auto disarm if lockdown

 - auto disarm when locked down was added in #14766 to prevent user
confusion in regular usage, but also breaks HITL where lockdown is
enabled for safety
 - fixes #15686
commit 99b8d422 (2020-09-06): commander: HITL skip auto disarm if lockdown

 - auto disarm when locked down was added in #14766 to prevent user
confusion in regular usage, but also breaks HITL where lockdown is
enabled for safety
 - fixes #15686
PR #15690 (2020-09-06): commander: HITL skip auto disarm if lockdown
 - auto disarm when locked down was added in #14766 to prevent user
confusion in regular usage, but also breaks HITL where lockdown is
enabled for safety
 - fixes #15686

I'd actually like to purge most of these HITL special cases in commander, but not today.
commit 1ac31100 (2020-09-06): commander: HITL skip auto disarm if lockdown

 - auto disarm when locked down was added in #14766 to prevent user confusion in regular usage, but also breaks HITL where lockdown is enabled for safety
 - fixes #15686
closed 2020-09-06 by dagar