PX4/PX4-Autopilot #15692: When Simulating JmavSim..Ubuntu 20.04 running on Raspberry Pi freezes

GokulS-dot, 2020-09-06:
when running "make px4_sitl jmavsim" inside Firmware directory
after sometimes ubuntu gets freezed and it has to reboot 
[image]

what's the issue??

hamishwillee, 2020-09-06:
Probably RasPi does not have enough power to handle a task like this. In any case, not enough information to debug.

GokulS-dot, 2020-09-07:
> Probably RasPi does not have enough power to handle a task like this. In any case, not enough information to debug.

But in Px4 Guide Manual...shows it's compatible with Raspberry Pi 2/3

hamishwillee, 2020-09-07:
Simulation is for development, so generally we'd expect a development computer - which RasPi is not. You can run PX4 on RasPi though as a flight controller or a companion computer.

GokulS-dot, 2020-09-07:
> Simulation is for development, so generally we'd expect a computer with enough power for development - which RasPi is not. You can run PX4 on RasPi though as a flight controller or a companion computer.

When I run stimulation on Debian Linux in RPI..everything gets downloaded , got error for the very last step ☹️
FAILED:Linking bin/Px4
/use/include/c++/8/bits/atomic_base.h:396: error :undefined reference to '_atomic_load_8'

hamishwillee, 2020-09-08:
Again, using an underpowered RPi as a development machine does not seem wise. That said, the times I have had errors like this it has usually be either incorrect installation or incorrect submodule setup. Try the easy one first:
```
make clean
git submodule update --recursive
make px4_sitl jmavsim
```
If that does not work, not sure. @julianoes Thoughts?

GokulS-dot, 2020-09-08:
> Again, using an underpowered RPi as a development machine does not seem wise. That said, the times I have had errors like this it has usually be either incorrect installation or incorrect submodule [...]

@julianoes any issue with Raspbian gcc compiler ??