PX4/PX4-Autopilot #15699: How to link the library

maoyexin, 2020-09-08:
make px4_sitl gazebo        
[0/5] Performing build step for 'sitl_gazebo'
[2/98] Linking CXX shared library libmav_msgs.dylib
FAILED: libmav_msgs.dylib 
: && /Library/Developer/CommandLineTools/usr/bin/c++ -O2 -g -DNDEBUG -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk -dynamiclib -Wl,-headerpad_max_install_names  -o libmav_msgs.dylib -install_name @rpath/libmav_msgs.dylib CMakeFiles/mav_msgs.dir/CommandMotorSpeed.pb.cc.o CMakeFiles/mav_msgs.dir/MotorSpeed.pb.cc.o -L/usr/local/Cellar/gazebo11/11.1.0/lib   -L/usr/local/Cellar/gazebo11/11.1.0/lib/gazebo-11/plugins   -L/usr/local/Cellar/bullet/2.89/lib   -L/Users/myx/Firmware/build/px4_sitl_default/build_gazebo   -L/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_8/lib -Wl,-rpath,/usr/local/Cellar/gazebo11/11.1.0/lib -Wl,-rpath,/usr/local/Cellar/gazebo11/11.1.0/lib/gazebo-11/plugins -Wl,-rpath,/usr/local/Cellar/bullet/2.89/lib -Wl,-rpath,/Users/myx/Firmware/build/px4_sitl_default/build_gazebo -Wl,-rpath,/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_8/lib -Wl,-rpath,/usr/local/lib  /usr/local/lib/libboost_system-mt.dylib  /usr/local/lib/libboost_thread-mt.dylib  -lBulletSoftBody  -lBulletDynamics  -lBulletCollision  -lLinearMath  /usr/local/lib/libSimTKsimbody.3.7.dylib  /usr/local/lib/libdart.6.9.4.dylib  /usr/local/Cellar/gazebo11/11.1.0/lib/libgazebo.dylib  /usr/local/Cellar/gazebo11/11.1.0/lib/libgazebo_client.dylib  /usr/local/Cellar/gazebo11/11.1.0/lib/libgazebo_gui.dylib  /usr/local/Cellar/gazebo11/11.1.0/lib/libgazebo_sensors.dylib  /usr/local/Cellar/gazebo11/11.1.0/lib/libgazebo_rendering.dylib  /usr/local/Cellar/gazebo11/11.1.0/lib/libgazebo_physics.dylib  /usr/local/Cellar/gazebo11/11.1.0/lib/libgazebo_ode.dylib  /usr/local/Cellar/gazebo11/11.1.0/lib/libgazebo_transport.dylib  /usr/local/Cellar/gazebo11/11.1.0/lib/libgazebo_msgs.dylib  /usr/local/Cellar/gazebo11/11.1.0/lib/libgazebo_util.dylib  /usr/local/Cellar/gazebo11/11.1.0/lib/libgazebo_common.dylib  /usr/local/Cellar/gazebo11/11.1.0/lib/libgazebo_gimpact.dylib  /usr/local/Cellar/gazebo11/11.1.0/lib/libgazebo_opcode.dylib  /usr/local/Cellar/gazebo11/11.1.0/lib/libgazebo_opende_ou.dylib  /usr/local/lib/libboost_system-mt.dylib  /usr/local/lib/libboost_filesystem-mt.dylib  /usr/local/lib/libboost_program_options-mt.dylib  /usr/local/lib/libboost_regex-mt.dylib  /usr/local/lib/libboost_iostreams-mt.dylib  /usr/local/lib/libprotobuf.dylib  "/usr/local/lib/libsdformat9.9.3.0~pre1.dylib"  /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_8/lib/libOgreMain.dylib  /usr/local/lib/libboost_thread-mt.dylib  /usr/local/lib/libboost_date_time-mt.dylib  /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_8/lib/libOgreTerrain.dylib  /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_8/lib/libOgrePaging.dylib  /usr/local/lib/libignition-common3-graphics.3.6.0.dylib  /usr/local/lib/libopencv_gapi.4.4.0.dylib  /usr/local/lib/libopencv_stitching.4.4.0.dylib  /usr/local/lib/libopencv_alphamat.4.4.0.dylib  /usr/local/lib/libopencv_aruco.4.4.0.dylib  /usr/local/lib/libopencv_bgsegm.4.4.0.dylib  /usr/local/lib/libopencv_bioinspired.4.4.0.dylib  /usr/local/lib/libopencv_ccalib.4.4.0.dylib  /usr/local/lib/libopencv_dnn_objdetect.4.4.0.dylib  /usr/local/lib/libopencv_dnn_superres.4.4.0.dylib  /usr/local/lib/libopencv_dpm.4.4.0.dylib  /usr/local/lib/libopencv_face.4.4.0.dylib  /usr/local/lib/libopencv_freetype.4.4.0.dylib  /usr/local/lib/libopencv_fuzzy.4.4.0.dylib  /usr/local/lib/libopencv_hfs.4.4.0.dylib  /usr/local/lib/libopencv_img_hash.4.4.0.dylib  /usr/local/lib/libopencv_intensity_transform.4.4.0.dylib  /usr/local/lib/libopencv_line_descriptor.4.4.0.dylib  /usr/local/lib/libopencv_quality.4.4.0.dylib  /usr/local/lib/libopencv_rapid.4.4.0.dylib  /usr/local/lib/libopencv_reg.4.4.0.dylib  /usr/local/lib/libopencv_rgbd.4.4.0.dylib  /usr/local/lib/libopencv_saliency.4.4.0.dylib  /usr/local/lib/libopencv_sfm.4.4.0.dylib  /usr/local/lib/libopencv_stereo.4.4.0.dylib  /usr/local/lib/libopencv_structured_light.4.4.0.dylib  /usr/local/lib/libopencv_superres.4.4.0.dylib  /usr/local/lib/libopencv_surface_matching.4.4.0.dylib  /usr/local/lib/libopencv_tracking.4.4.0.dylib  /usr/local/lib/libopencv_videostab.4.4.0.dylib  /usr/local/lib/libopencv_viz.4.4.0.dylib  /usr/local/lib/libopencv_xfeatures2d.4.4.0.dylib  /usr/local/lib/libopencv_xobjdetect.4.4.0.dylib  /usr/local/lib/libopencv_xphoto.4.4.0.dylib  /usr/local/lib/libprotobuf.dylib  /usr/local/Cellar/gstreamer/1.16.2/lib/libgstreamer-1.0.dylib  -lglib-2.0  -lgobject-2.0  /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_8/lib/libOgreMain.dylib  /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_8/lib/libOgreTerrain.dylib  /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_8/lib/libOgrePaging.dylib  /usr/local/Cellar/gstreamer/1.16.2/lib/libgstreamer-1.0.dylib  -lglib-2.0  -lgobject-2.0  /usr/local/lib/libSimTKmath.3.7.dylib  /usr/local/lib/libSimTKcommon.3.7.dylib  -framework  Accelerate  -ldl  -lm  /usr/local/lib/libdart-external-odelcpsolver.6.9.4.dylib  /usr/local/lib/libccd.2.0.dylib  /usr/lib/libm.dylib  /usr/local/Cellar/fcl/0.6.1/lib/libfcl.dylib  /usr/local/Cellar/assimp/5.0.1/lib/libassimp.dylib  /usr/local/lib/liboctomap.1.9.5.dylib  /usr/local/lib/liboctomath.1.9.5.dylib  /usr/local/lib/libignition-transport8.8.1.0.dylib  /usr/local/lib/libignition-fuel_tools4.4.2.1.dylib  /usr/local/lib/libignition-msgs5.5.3.0.dylib  /usr/local/lib/libignition-math6.6.4.0.dylib  /usr/local/lib/libprotobuf.dylib  /usr/local/lib/libignition-common3.3.6.0.dylib  /usr/local/Cellar/ossp-uuid/1.6.2_2/lib/libuuid.dylib  /usr/local/Cellar/ossp-uuid/1.6.2_2/lib/libuuid.dylib  -lpthread  /usr/local/lib/libopencv_highgui.4.4.0.dylib  /usr/local/lib/libopencv_shape.4.4.0.dylib  /usr/local/lib/libopencv_datasets.4.4.0.dylib  /usr/local/lib/libopencv_plot.4.4.0.dylib  /usr/local/lib/libopencv_text.4.4.0.dylib  /usr/local/lib/libopencv_dnn.4.4.0.dylib  /usr/local/lib/libopencv_ml.4.4.0.dylib  /usr/local/lib/libopencv_phase_unwrapping.4.4.0.dylib  /usr/local/lib/libopencv_optflow.4.4.0.dylib  /usr/local/lib/libopencv_ximgproc.4.4.0.dylib  /usr/local/lib/libopencv_video.4.4.0.dylib  /usr/local/lib/libopencv_videoio.4.4.0.dylib  /usr/local/lib/libopencv_imgcodecs.4.4.0.dylib  /usr/local/lib/libopencv_objdetect.4.4.0.dylib  /usr/local/lib/libopencv_calib3d.4.4.0.dylib  /usr/local/lib/libopencv_features2d.4.4.0.dylib  /usr/local/lib/libopencv_flann.4.4.0.dylib  /usr/local/lib/libopencv_photo.4.4.0.dylib  /usr/local/lib/libopencv_imgproc.4.4.0.dylib  /usr/local/lib/libopencv_core.4.4.0.dylib && :
ld: library not found for -lglib-2.0
clang: error: linker command failed with exit code 1 (use -v to see invocation)
[3/98] Running cpp protocol buffer compiler on msgs/Groundtruth.proto
ninja: build stopped: subcommand failed.
[1/5] Generating ../../logs
FAILED: external/Stamp/sitl_gazebo/sitl_gazebo-build 
cd /Users/myx/Firmware/build/px4_sitl_default/build_gazebo && /usr/local/Cellar/cmake/3.18.2/bin/cmake --build /Users/myx/Firmware/build/px4_sitl_default/build_gazebo -- -j2
ninja: build stopped: subcommand failed.
make: *** [px4_sitl] Error 1

Jaeyoung-Lim, 2020-09-08:
@maoyexin Which system are you running this in? probably macOS?

maoyexin, 2020-09-08:
Yes, macOs.
It looks like i can't link  libmav_msgs.dylib.

maoyexin, 2020-09-08:
> @maoyexin Which system are you running this in? probably macOS?

Yes, macOs.
It looks like i can't link libmav_msgs.dylib.

rosshettel, 2020-10-19:
I am also seeing this issue on macOS. Running `pkg-config --libs glib-2.0 ` did not solve it

mikeklinker129, 2020-12-23:
I am running Mac Catalina 10.15.6 and I see the same error. Any updates?

canberkdurmus, 2020-12-27:
Any updates on this issue? I am completely stuck with my development environment. Catalina 10.15.7

LorenzMeier, 2020-12-29:
This is all working on master (I run two systems, one on Catalina and one on Big Sur). Please make sure to:
1. Have an up-to-date Homebrew installation
2. Upgrade to current master of PX4 to verify
3. Run `make distclean` in the PX4 folder

Please report back if it does work for you or not.

canberkdurmus, 2020-12-29:
Hello, the problem is solved for me at [#679](https://github.com/PX4/PX4-SITL_gazebo/issues/679)

Running Catalina 10.15.7
Homebrew 2.7.1-27
Command Line Tools: 12.3.0.0.1.1607026830
Autopilot Master Branch

LorenzMeier, 2020-12-29:
Perfect, thanks for the detailed report! I'm closing this accordingly. Please don't hesitate to reach out if you encounter more issues.

closed 2020-12-29 by LorenzMeier