PX4/PX4-Autopilot #15701: Battery readings - always 100% on px4 1.11.0

xenojimmy, 2020-09-08:
dear coomunity,
i'm facing an issue regarding the battery reading, I always get 100% even when the voltage is far below that value.
conditions:
**- pixhawk 2.1 cube
- version 1.11.0
- bat_V_div calibrated using an external multimeter.
- default max min voltage
- exact number of cells**

according to the documentation, in this new version the multi battery option gives the chance to use both indipendently 
**- bat_**  (deprecated)
**- bat1_
- bat2_**

what I notice is that the behavior is different if u use the power configurator (it acts only on the deprecated variables) or if u use the parameter list varying **bat1_** values...

this is all  I got by troubleshooting the system, another pixhawk 2.1 using 1.10.0 doesn't give such problems, it uses onyl **bat_** variables

thank you, for your support

modaltb, 2020-09-09:
I just observed this as well on modalai_fc-v1 target.  Looks like setting BAT1_N_CELLS did the trick for me to get proper percentage going.

moreba1, 2020-09-27:
I have this problem with pixhawk 2.1 cube  in 1.11 version, voltage of the battery did not change with calibration.

igalloway, 2020-10-26:
For future reference - this is also the case for NXP RDDRONE-FMUK66

YangZJ2000m, 2024-09-29:
Greetings, my friends. I've encountered exactly the same problem four years after your post, and I have managed to solve it with a stupid(but simple) modification. 

**Flight Platform**
- 12S Cell battery
- CUAV_V5+ Flight Control
- Version 1.11.2

**Question Description**
Same as the posted content: I've correctly set the number of cells, maximum voltage, and minimum voltage, and calculated the divided voltage accordingly. However, the display of battery percentage on QGroundControl is constantly 100%, which is obviously a mistake.

**Problem Analysis**
This problem is caused by version 1.11, which can't display the correct battery percentage for batteries with more than 10S cells. The code causing this problem is in the file battery_status.h and battery.cpp. In short, the voltage of each cell is wrongly calculated and wrongly assigned.

[image]
[image]

For batteries with no more than 10S cells, there would not be any trouble.
For batteries with more than 10S cells, version 1.11 would evenly set voltage_cell_v[10] to 'actual voltage / actual cells', resulting in the loss of voltage readings for the extra cells. For example:

I'm using a 12S battery with a measured actual voltage of 45.45V. After performing the voltage calibration as shown in the following picture (this would wrongly set the BAT1_V_DIV value, causing even more trouble).

[image]

 QGroundControl would show the correct voltage of 45.44V, but the battery percentage is incorrect (constantly 100%). Mavlink messages showed the following: battery_status reports voltages = [4544, 4544, 4544, 4544, 4544, 4544, 4544, 4544, 4544, 4544] (unit: mV, which is obviously wrong because a charged cell voltage would be no more than 4.2V), while sys_status reports battery_voltage = 54529 (unit: mV).

[image]
[image]

The math is as follows: 54.529V / 12 = 4.544V. Normally, the voltage of the 12 cells would be calculated as 4.544V * 12; however, according to the code mentioned above, the voltage is set to 4.544V * 10 = 45.44V. This incorrect calculation leads to a correct voltage display, but in a wrong way. Now the system recorded voltage is 54.529V, which is way higher than the maximum voltage of 4.2V * 12 = 50.4V. Therefore, the display of battery level is 100%, which is wrong because the actual level is currently 43%.

**Possible Solution**
This problem is solved in later versions. However, it can also be addressed with a stupid but simple modification in the code, as follows. 

[image]

The logic of this modification is: since the code supports no more than 10 cells, I've added logic to check if the number of cells is more than 10. If the number of cells ≤ 10, use the old code; there would be no trouble at all. Else, if the number of cells > 10, considering that the calculation would only assign the voltage of 10 cells, the denominator would be fixed at 10 rather than the number of cells.

As a result (after re-calibration):
- Correct voltage of the battery
- Correct percentage of the battery (Hooray! Problem solved)
- Incorrect voltage for each cell (No big deal, since version 1.11 would also assign the wrong value)

closed 2026-04-10 by github-actions[bot]