PX4/PX4-Autopilot #15703: Error:: Java Error pops up when jmavsim_run.sh file executed!!

GokulS-dot, 2020-09-08:
OS : Ubuntu 20.04.1
Device : Raspberry Pi4
Issues : Under Firmware Directory,
             cd Firmware
             bash ./Tools/jmavsim_run.sh
             
[image]
java version :
openjdk 14.0.1 2020-04-14
OpenJDK Runtime Environment (build 14.0.1+7-Ubuntu-1ubuntu1)
OpenJDK 64-Bit Server VM (build 14.0.1+7-Ubuntu-1ubuntu1, mixed mode, sharing)

//////****How to Solve this issue*********////////

hamishwillee, 2020-09-08:
Did you set the environment using the build script ubuntu.sh in https://dev.px4.io/master/en/setup/dev_env_linux_ubuntu.html ?

GokulS-dot, 2020-09-09:
> Did you set the environment using the build script ubuntu.sh in https://dev.px4.io/master/en/setup/dev_env_linux_ubuntu.html ?

Yes...set the environment

julianoes, 2020-09-09:
It looks like the dependency libgluegen_rt.jar does not work on aarch64. When I updated the dependencies I did not copy the latest for aarch64, see https://github.com/PX4/jMAVSim/pull/119/files.

You could try to update the libs for aarch64 from here: https://jogamp.org/deployment/jogamp-next/jar/

Note that in general Raspberry Pi as a host platform is not something that we support (yet).

GokulS-dot, 2020-09-09:
> It looks like the dependency libgluegen_rt.jar does not work on aarch64. When I updated the dependencies I did not copy the latest for aarch64, see https://github.com/PX4/jMAVSim/pull/119/files. [...]

I will try with that....
I could have a doubt !!
I have Px4 running on one computer of some ip and jmav sim on other system of other ip..both are on same network..can it be connect ..I couldnt connect

GokulS-dot, 2020-09-09:
> It looks like the dependency libgluegen_rt.jar does not work on aarch64. When I updated the dependencies I did not copy the latest for aarch64, see https://github.com/PX4/jMAVSim/pull/119/files. [...]

[image]
Finally jmavsim works on RPI...
Great Tanxx Man!!
Has to load jogl aarch64version also...worked finally 😊

GokulS-dot, 2020-09-09:
> I will try with that....
> I could have a doubt !!
> I have Px4 running on one computer of some ip and jmav sim on other system of other ip..both are on same network..can it be connect ..I couldnt connect

@julianoes

julianoes, 2020-09-16:
Nice! Could you make a PR where these files are copied in for the next one that tries it on RPi?

GokulS-dot, 2020-09-16:
> Nice! Could you make a PR where these files are copied in for the next one that tries it on RPi?

PR??