PX4/PX4-Autopilot #15708: Support thrust setpoints for the hippocampus vehicle in mc_rate_control

lennartalff, 2020-09-09:
For general questions please use [PX4 Discuss](http://discuss.px4.io/) or [Slack](http://slack.px4.io/).

**Describe problem solved by the proposed feature**
I would like to use the `mc_rate_control` module for the uuv_hippocampus vehicle. The problem I'm facing is that the mavlink message [SET_ATTITUDE_TARGET](https://mavlink.io/en/messages/common.html#SET_ATTITUDE_TARGET) defines thrust as scalar. The `mavlink_receiver` module uses the `fill_thrust`-method to fill the thrust vector used in uORB. For `MAV_TYPE_SUBMARINE` this scalar thrust is written to the x-component (`thrust[0]`) of the thrust vector.

`mc_rate_control` sets the thrust to `thrust[2]`. But the thrust set by the SET_ATTITUDE_TARGET message is in `thrust[0]`, so it is not possible to control the thrust.

**Describe your preferred solution**
Check the `MAV_TYPE` in the `mc_rate_control` module and chose the appropriate thrust component accordingly.

**Describe possible alternatives**
Check which thrust component is different from zero and select this for the thrust output. If all components are zero, the output should be 0 anyway.

**Additional context**
My current workaround is to set the MAV_TYPE to helicopter so the `mavlink_receiver` writes the thrust value to `thrust[2]`.

Jaeyoung-Lim, 2020-12-07:
@lennartalff Are there any updates regarding this issue?

Jaeyoung-Lim, 2021-01-17:
Closing as stale

closed 2021-01-17 by Jaeyoung-Lim