PX4/PX4-Autopilot #15711: Rover no longer supported?

nikored, 2020-09-10:
I've loaded the lastest Firmware version (1.11.0) in a Pixhawk board and, when I tried to choose a "Rover" frame, the option did not show up. So I followed the instructions in https://discuss.px4.io/t/airframes-rover-bug/6267/3 and deleted the PX4AirframeFactMetaData.xml file in C:\Users\User\AppData\Roaming\QGroundControl.org
It now shows the option to select a "Rover" frame, but when I select it and click apply and restart, the board does restart but nothing else happens, the software still shows me the following image, as if I did not choose an airframe:
[image]
I tried with other versions of the 1.11.0 firmware too, such as the beta testing and developer build, and had the same result.
I even tried to manually load the px4_fmu-v2_rover.px4 file and had the same result.

When I use an earlier version (i.e. 1.10.0) I don't have this problem:
[image]
Also, when I select another airframe such as a generic quadcopter x, it gets selected without any problem:
[image]

Any Idea how to fix this? Or should I give up on using the 1.11 version?

Jaeyoung-Lim, 2020-09-10:
> I've loaded the lastest Firmware version (1.11.0) in a Pixhawk board and, when I tried to choose a "Rover" frame, the option did not show up. 

Which pixhawk board are you using?

nikored, 2020-09-10:
> > I've loaded the lastest Firmware version (1.11.0) in a Pixhawk board and, when I tried to choose a "Rover" frame, the option did not show up.
> 
> Which pixhawk board are you using?

A Pixhawk 1 board

hamishwillee, 2020-09-10:
You're right it was removed for FMUv2. It is possible you have a Pixhawk 1 that can be upgraded to use FMUv3 - that is the first thing to try: http://docs.px4.io/master/en/config/firmware.html#bootloader

@dagar Is it possible to get Rover on FMUv2? If so, how?

nikored, 2020-09-10:
I tried to upgrade the bootloader but I couldn't. I guess the Pixhawk 1 is an early version.
I'll just use the 1.10 version meanwhile. Thank you for answering me.

dagar, 2020-09-13:
There's a dedicated `px4_fmu-v2_rover` build target, but it now occurs to me that the airframe pruning mechanism at build time applies to the board (px4_fmu-v2), not the variant (px4_fmu-v2_rover). https://github.com/PX4/Firmware/blob/ac732cdebad261f0008ebcadaf78e0afbee29716/ROMFS/px4fmu_common/init.d/airframes/50000_generic_ground_vehicle#L8

hamishwillee, 2020-09-15:
@dagar So would the short term solution be to modify the source code to remove that line and build px4_fmu-v2_rover yourself? 
Are you planning to fix this properly?

hamishwillee, 2020-09-28:
@dagar ^^^

dagar, 2020-09-29:
Is generic rover (50000_generic_ground_vehicle) sufficient? If that's all then yes let's remove the exclude so that px4_fmu-v2_rover can function. I have some ideas for a better overall airframe inclusion (and organization) method, but that will take longer.

dagar, 2020-09-29:
https://github.com/PX4/Firmware/pull/15849

berndpfrommer, 2020-11-04:
I tried installing the latest firmware from the "master" branch today via qgroundcontrol onto an old pixhawk 2.4.8 (px4 fmu v2). The firmware should have the fix now, but alas, I still cannot switch to the "generic ground vehicle" frame, with the exact same symptoms as described by the OP.

Jaeyoung-Lim, 2021-02-02:
Fixed in #15849

closed 2021-02-02 by Jaeyoung-Lim