PX4/PX4-Autopilot #15712: Build fails (v1.11.0, cubepilot_cubeorange_default)

Roman-, 2020-09-10:
Ubuntu 16.04 LTS, arm-none-eabi-gcc 6.3.1 20170620.
`make cubepilot_cubeorange_default upload` gives me this output:

[image]

This is the same issue as #15275 but for different board.

Roman-, 2020-09-10:
Same thing on Ubuntu 18.04 LTS.
`make cubepilot_cubeorange_default upload`
```
...
[ 93%] Building C object src/systemcmds/tests/CMakeFiles/systemcmds__tests.dir/test_param.c.obj
[ 93%] Built target drivers__imu__adis16448
Scanning dependencies of target drivers__imu__adis16497
[ 93%] Building CXX object src/systemcmds/tests/CMakeFiles/systemcmds__tests.dir/test_parameters.cpp.obj
[ 94%] Building CXX object src/drivers/imu/adis16497/CMakeFiles/drivers__imu__adis16497.dir/ADIS16497.cpp.obj
[ 94%] Building CXX object src/drivers/imu/adis16497/CMakeFiles/drivers__imu__adis16497.dir/adis16497_main.cpp.obj
[ 94%] Linking CXX static library libdrivers__imu__adis16477.a
[ 94%] Built target drivers__imu__adis16477
[ 94%] Building C object src/systemcmds/tests/CMakeFiles/systemcmds__tests.dir/test_perf.c.obj
Scanning dependencies of target drivers__imu__invensense__icm20602
[ 94%] Building CXX object src/systemcmds/tests/CMakeFiles/systemcmds__tests.dir/test_ppm.cpp.obj
[ 94%] Building CXX object src/drivers/imu/invensense/icm20602/CMakeFiles/drivers__imu__invensense__icm20602.dir/ICM20602.cpp.obj
[ 94%] Building CXX object src/drivers/imu/invensense/icm20602/CMakeFiles/drivers__imu__invensense__icm20602.dir/icm20602_main.cpp.obj
[ 94%] Linking CXX static library libdrivers__imu__adis16497.a
[ 94%] Building C object src/systemcmds/tests/CMakeFiles/systemcmds__tests.dir/test_ppm_loopback.c.obj
[ 94%] Built target drivers__imu__adis16497
[ 94%] Building C object src/systemcmds/tests/CMakeFiles/systemcmds__tests.dir/test_rc.c.obj
In file included from /home/user/repos/Firmware/platforms/common/include/px4_platform_common/atomic_bitset.h:36:0,
                 from /home/user/repos/Firmware/src/modules/uORB/uORBDeviceMaster.hpp:54,
                 from /home/user/repos/Firmware/src/modules/uORB/uORBDeviceNode.hpp:37,
                 from /home/user/repos/Firmware/src/modules/uORB/Publication.hpp:45,
                 from /home/user/repos/Firmware/src/modules/uORB/PublicationMulti.hpp:45,
                 from /home/user/repos/Firmware/src/lib/drivers/accelerometer/PX4Accelerometer.hpp:40,
                 from /home/user/repos/Firmware/src/drivers/imu/invensense/icm20602/ICM20602.hpp:46,
                 from /home/user/repos/Firmware/src/drivers/imu/invensense/icm20602/ICM20602.cpp:34:
/home/user/repos/Firmware/platforms/common/include/px4_platform_common/atomic.h: In instantiation of 'bool px4::atomic<T>::compare_exc$
ange(T*, T) [with T = unsigned int]':
/home/user/repos/Firmware/src/drivers/imu/invensense/icm20602/ICM20602.cpp:406:76:   required from here
/home/user/repos/Firmware/platforms/common/include/px4_platform_common/atomic.h:169:34: error: parameter 'expected' set but not used [$
Werror=unused-but-set-parameter]
  inline bool compare_exchange(T *expected, T desired)
                                  ^~~~~~~~
compilation terminated due to -Wfatal-errors.
cc1plus: all warnings being treated as errors
src/drivers/imu/invensense/icm20602/CMakeFiles/drivers__imu__invensense__icm20602.dir/build.make:62: recipe for target 'src/drivers/im$
/invensense/icm20602/CMakeFiles/drivers__imu__invensense__icm20602.dir/ICM20602.cpp.obj' failed
make[4]: *** [src/drivers/imu/invensense/icm20602/CMakeFiles/drivers__imu__invensense__icm20602.dir/ICM20602.cpp.obj] Error 1
make[4]: *** Waiting for unfinished jobs....

...

CMakeFiles/Makefile2:18288: recipe for target 'platforms/nuttx/CMakeFiles/upload.dir/rule' failed
make[2]: *** [platforms/nuttx/CMakeFiles/upload.dir/rule] Error 2
Makefile:3865: recipe for target 'upload' failed
make[1]: *** [upload] Error 2
Makefile:217: recipe for target 'cubepilot_cubeorange_default' failed
make: *** [cubepilot_cubeorange_default] Error 2

```

skystalkerrc, 2020-11-11:
This issue can be solved by commenting out the -Werror in file cmake/px4_add_common_flags.cmake.

farhangnaderi, 2021-02-22:
This solved my problem as well.

LorenzMeier, 2021-02-22:
This is fixed on master.

closed 2021-02-22 by LorenzMeier