PX4/PX4-Autopilot #15718: Thrust in Z set to zero with ground contact

parakhm95, 2020-09-11:
**Description**
We were flying our small hexarotor with a payload of 1.8 kgs and aircraft weight of 2.7 bringing total to 4.6 kgs. When touching the load down, I noticed a reduction in thrust and I could tell it was the land detector. But the surprising thing is that the ground_contact state is triggering zero thrust in Z axis when it should only set x and y axis to zero. Upon further inspection, I found that in v.1.10.2, on line 964 is this : 

```
void
MulticopterPositionControl::limit_thrust_during_landing(vehicle_local_position_setpoint_s &setpoint)
{
	if (_vehicle_land_detected.ground_contact
	    || _vehicle_land_detected.maybe_landed) {
		// we set thrust to zero, this will help to decide if we are actually landed or not
		setpoint.thrust[0] = setpoint.thrust[1] = setpoint.thrust[2] = 0.0f;
		// set yaw-sp to current yaw to avoid any corrections
		setpoint.yaw = _states.yaw;
		setpoint.yawspeed = 0.f;
		// prevent any integrator windup
		_control.resetIntegralXY();
		_control.resetIntegralZ();
	}
}
```

**Expected behavior**
According to the documentation, and also the logic that I can form in my head, this is either incorrect logic or a bug. Also, the docs mention the same seemingly correct logic here : https://docs.px4.io/v1.10/en/advanced_config/land_detector.html

**Log Files and Screenshots**
https://review.px4.io/plot_app?log=56814a95-d760-4d08-b840-280a69869811
[image]

**Cargo drones**
We fly cargo drones with 100 kgs payload capacity and we are slowly transitioning to PX4. This questions the general expectations from our side on what we can achieve on this platform. Could you throw some light on this too please?

parakhm95, 2020-09-11:
https://github.com/PX4/Firmware/commit/5887a2393cefad7268d165e07352384be61e1404#diff-fc77c3ef569029d45764664c75d4b0c1

This seems to be where author made the change but then 

https://github.com/PX4/Firmware/commit/d5903853dfa1a990c8f216a4113ec521970b34db#diff-fc77c3ef569029d45764664c75d4b0c1

this one merged them. I am unsure if this is an error anymore. It looks to be desired and willed. Maybe I am missing something.

dagar, 2020-09-13:
The current of `ground_contact` causing the multicopter position controller to drop the thrust is intentional, but it's been made slightly more robust in v1.11 with tightening the conditions. There's also been discussion of `ground_contact` causing the controller to try gradually reducing the thrust.

Can you elaborate on the core problem you're having? I'm not exactly sure what you mean by "when touching the payload", is this not during a regular landing?

FYI @MaEtUgR

parakhm95, 2020-09-14:
I realize that my description is incomplete. Our drones primarily carry swung payload weighing upto 100 kgs. The payload is connected to a load hook which can be released upon touchdown. The videos below are short and will help in illustrating my point better : 

https://www.youtube.com/watch?v=4e7EV70m8Dw
https://www.youtube.com/watch?v=GWl2qlf-T0k

What are these tighter conditions that you mention here for v1.11? My logic for landing would be about detecting velocity limit first and checking if the aircraft is not moving when given a setpoint and then using the low thrust conditions later on after further reduction of the thrust which will happen because of I-term wind-up if velocity setpoint cannot be reached. This should be to handle wind updraft situations too because that will trigger the no motion flag and the low thrust flag but not the minimal thrust and not the velocity setpoint. But again, I might be missing some edge case here. 

And yes, so we want to be able to place our payloads down, then detach the tether/rope from the hook and then return to take-off point. The videos attached above will help illustrate my point.

parakhm95, 2020-09-28:
Hey folks, I think our conversation has gone stale and so here's a reminder to breathe new life into it.

dagar, 2020-09-28:
> My logic for landing would be about detecting velocity limit first and checking if the aircraft is not moving when given a setpoint and then using the low thrust conditions later on after further [...]

Do you want to open a pull result with test results we can review in detail and discuss? https://github.com/PX4/Firmware/blob/bb77f55f7b59c71b2edcc5752db71d4db001a2ca/src/modules/land_detector/MulticopterLandDetector.cpp#L153-L230