PX4/PX4-Autopilot #15719: make px4_fmu-v2_default Error

oscartiantian, 2020-09-12:
I install Development Environment on Ubuntu 18.04 LTS with ubuntu.sh and ubuntu_sim_ros_melodic.sh，（arm-none-eabi-gcc 9.3.1 20200408），when make px4_fmu-v2_default  , gives output:
-- PX4 version: v1.11.0-rc3-46-g5c040582db
-- PX4 config file: /home/mrgao/PX4/Firmware/boards/px4/fmu-v2/default.cmake
-- PX4 config: px4_fmu-v2_default
-- PX4 platform: nuttx
/usr/bin/env: ‘bash\r’: No such file or directory
/usr/bin/env: ‘bash\r’: No such file or directory
make[1]: *** [include/arch] Error 127
-- cmake build type: MinSizeRel
-- The CXX compiler identification is unknown
-- The C compiler identification is unknown
-- The ASM compiler identification is unknown
-- Found assembler: arm-none-eabi-gcc
CMake Error at CMakeLists.txt:202 (project):
  The CMAKE_CXX_COMPILER:

    arm-none-eabi-g++

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.

CMake Error at CMakeLists.txt:202 (project):
  The CMAKE_C_COMPILER:

    arm-none-eabi-gcc

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "CC" or the CMake cache entry CMAKE_C_COMPILER to the full path to
  the compiler, or to the compiler name if it is in the PATH.

CMake Error at CMakeLists.txt:202 (project):
  The CMAKE_ASM_COMPILER:

    arm-none-eabi-gcc

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "ASM" or the CMake cache entry CMAKE_ASM_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.

-- Warning: Did not find file Compiler/-ASM
-- Configuring incomplete, errors occurred!
See also "/home/mrgao/PX4/Firmware/build/px4_fmu-v2_default/CMakeFiles/CMakeOutput.log".
See also "/home/mrgao/PX4/Firmware/build/px4_fmu-v2_default/CMakeFiles/CMakeError.log".
Error: /home/mrgao/PX4/Firmware/build/px4_fmu-v2_default is not a directory
Makefile:217: recipe for target 'px4_fmu-v2_default' failed
make: *** [px4_fmu-v2_default] Error 1

But arm-none-eabi-gcc --version , gives:
arm-none-eabi-gcc (GNU Arm Embedded Toolchain 9-2020-q2-update) 9.3.1 20200408 (release)
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Jaeyoung-Lim, 2020-12-07:
You are missing a dependency
```
apt install arm-none-eabi-gcc arm-none-eabi-newlib
```

closed 2020-12-07 by Jaeyoung-Lim

HadesGuy, 2022-10-21:
how do you install this dependency in Nuttx tho ?

mnumanuyar, 2023-06-08:
ok so, recently i got this error and solution did not work. while searching, i foun fallowing; is this relevant: https://askubuntu.com/questions/1243252/how-to-install-arm-none-eabi-gdb-on-ubuntu-20-04-lts-focal-fossa

xianglunkai, 2023-06-08:
if you are using pixhawk-v2， I suggest make px4_fmu-v3_default  instead.

mnumanuyar, 2023-06-08:
what solved for me is : `sudo apt -y install gcc-arm-none-eabi`