PX4/PX4-Autopilot #15725: Power redundancy not met: 0 instead of 1

neo0605-d, 2020-09-13:
Hello. Have installed firmware 1.11.0. Conducted calibration of sensors, radios, motors. Does not allow switching on motors (arm), error: power redundancy not met: instead of 1
[image]

dagar, 2020-09-13:
Which flight controller and how is it powered?

neo0605-d, 2020-09-13:
Pixhawk 2.4.8 controller, powered by a power distribution Board.
[image]

dagar, 2020-09-13:
> Pixhawk 2.4.8 controller, powered by a power distribution Board.
> [image]

If you can capture a log from boot (SDLOG_MODE) and share it https://logs.px4.io/ I'll take a closer look. Make sure the log includes attempting to arm.

neo0605-d, 2020-09-13:
Thank you in advance. I'll do it tomorrow. Good time of day.

ryanjAA, 2020-09-13:
@neo0605-d set COM_POWER_COUNT to 0 from 1 and that warning will go away. This should either be set to 0 as standard or be board specific in the future.

dagar, 2020-09-14:
@ryanjAA it's counting the number of bricks, with the default set to 1 (no redundancy). Is it not actually working that way?

ryanjAA, 2020-09-14:
Interesting. No, it’s not working that way on v3 and v5 (pix1 and p4). Had to set to 0 to work although pix 4 testing will need to double check but I believe so. Definitely on pix 1.

ryanjAA, 2020-09-14:
Just checked (looked at logs). Works on pix 4. Not on pix 1.

neo0605-d, 2020-09-14:
Hello. Thank you very much for your help!!!!!!! Everything worked, I will test it on the street

closed 2020-09-14 by neo0605-d

YANG-DEMIN, 2020-09-15:
Hello, I meet the same problem. A few days ago,  i changed the battery capacity and calibrated the power module. When i change back the battery, the problem arise. After several times calibration, it did not work. 
Finally, **I flashed the Firmware, it worked again**.

ryanjAA, 2020-09-15:
You have this working with 1.11 and com_power_count set to 1? Which Pixhawk are you using? It works on Pixhawk 4 but not 1.

ssnake, 2020-09-16:
Firmware 1.11 and pixhawk v2 setting COM_POWER_COUNT to 0 helped. Thanks

paulr4, 2021-03-07:
I'm having this problem too, on a Pixhawk 2.4.8. The COM_POWER_COUNT = 0 suggestion worked for me, but shouldn't this ticket stay open until the fix suggested by @neo0605-d is implemented?

> @neo0605-d set COM_POWER_COUNT to 0 from 1 and that warning will go away. It’s just checking that you have a redundant power source. This should either be set to 0 as standard or be board specific in the future.