PX4/PX4-Autopilot #15732: Mpu9250 issue with Navio2 Firmware

GokulS-dot, 2020-09-14:
Instead of buying a Navio2 board..I download Navio2 Firmware to Raspberry Pi ..
Px4 installed correctly !!
Then I integerated GPS Neo 6M with Navio2 Firmware on RPI board..
That's also worked perfectly as I changed in Px4 config of GPS to uart as default gps in navio connects thrgh spi protocol
But Mpu9520 when connected , devices is not detected in Navio2 firmware I tried i2c config ..device not detected 

Is there any diff bw external mpu9520 and on board imu used in Navio2 board ??

dagar, 2020-09-14:
Some of the IMU drivers are still in flux because we have newer rewritten drivers that transfer all raw data (8 kHz gyro), but only support SPI. For now you could try using the old mpu9250 on i2c by changing the mpu9250 in the board config.

https://github.com/PX4/Firmware/blob/c15efea429faa9b5a3af802bbf0008eca956b762/boards/emlid/navio2/default.cmake#L24

 - old mpu9250 is in `src/drivers/imu/mpu9250`.
 - new mpu9250 is in `src/drivers/imu/invensense/mpu9250`.

Alternatively you could try this branch. https://github.com/PX4/Firmware/pull/15421

GokulS-dot, 2020-09-14:
> Some of the IMU drivers are still in flux because we have newer rewritten drivers that transfer all raw data (8 kHz gyro), but only support SPI. For now you could try using the old mpu9250 on i2c by [...]

Will try and check

GokulS-dot, 2020-09-14:
> Some of the IMU drivers are still in flux because we have newer rewritten drivers that transfer all raw data (8 kHz gyro), but only support SPI. For now you could try using the old mpu9250 on i2c by [...]

Inboard mpu9250 files in navio supports both i2c and spi?

dagar, 2020-09-14:
> Inboard mpu9250 files in navio supports both i2c and spi?

 * old mpu9250 is in `src/drivers/imu/mpu9250` (supports both i2c/spi, but only low rate data and sensor side filtering)
 * new mpu9250 is in `src/drivers/imu/invensense/mpu9250`. (currently SPI only, but there's an I2C version in https://github.com/PX4/Firmware/pull/15421)

closed 2021-01-17 by dagar