PX4/PX4-Autopilot #15733: Soft Reboot doesn't load specific parameter changes

ryanjAA, 2020-09-14:
Pixhawk 1 (FMU_v3)

1.11 Stable

When updating PWM_MAIN_MINx and PWM_MAIN_MAXx settings which require a reboot to take effect, they do not get pulled in from a soft reboot (from the tools/reboot vehicle button). Only with a hard power cycle.

Tested this multiple times. 

To recreate:

Change PWM_MAIN_MINx to something different (and PWM_MAIN_MAINx) and hit reboot, it doesn't actually institute the change even though the parameter is changed and lists as changed after the reboot. 

Photos:

Parameter changed and a soft Reboot (twice for good measure). Look at servo 6.

1400 us

[image]

Minimum specific output value is:

[image]

Overall minimum PWM output:
[image]

After a hard reboot (pulled the battery):

Goes to 1000 us

[image]

dagar, 2020-09-14:
That's weird, either way the same portion of init is executed regardless of reboot. 

Can you inspect the system a bit more directly through the console after the first reboot? 

`pwm info`
`param show PWM*`
`listener actuator_outputs`

ryanjAA, 2020-09-14:
Strange. It is working now but I saw this twice when writing today and pretty repeatedly on the test field. These are different aircraft. Testing had to stop to open the hatch for a power cycle 2 times throughout the day. Let me play with this more to recreate and capture it.

I'll keep looking at it or will close.

ryanjAA, 2020-09-14:
Here is it working after a soft reboot:

[image]
[image]
[image]
[image]
[image]

ryanjAA, 2020-09-15:
Ok, it's doing it again.

Did one soft reboot and here is what shows up. It didn't pull the min/max or trim back to 1500:

[image]
[image]
[image]
[image]
[image]
[image]

ryanjAA, 2020-09-15:
Just did another soft reboot and still the same.

dagar, 2020-09-16:
I don't think I'm following what you're doing exactly. Where is channel 6 getting min 1100 us and max 1700 us?

Although without following those details the difference is pretty clearly something with the px4io.

ryanjAA, 2020-09-16:
Guess I didn't hit comment... Retyping:

I am setting min and max manually to 1100/1700 just for that output channel. Then hard rebooting and all works. Then when setting those back to default (-1) so 1000 and 2000 min/max and soft rebooting (and after the soft reboot getting the console output above for you), nothing takes and it is holding the numbers from the previous state (hence the 1100/1700). I did another soft reboot on the last test as well after getting the output and it still stayed at 1100/1700. If I then hard reboot everything works as intended.

What is also strange is that this is not all the time. Most of the time but not all the time. Several different systems though have shown the same functionality (ruling out hardware itself). I don't recall seeing this on Pix4 but I couldn't say definitively since that setting was not altered as much during those flights so this is being done on a Pix1.