PX4/PX4-Autopilot #15734: Board serial number not exposed though USB serial port info

DonLakeFlyer, 2020-09-14:
QGC has support for only connecting to the first port of a composite USB device. This is done by keeping track of the vid/pid/serial numbers for each port connected to. This information comes from the USB hardware port info. When running PX4 firmware the serial number for all boards comes back as "0". ArduPilot Cube Orange support exposes this composite usb and also correctly reports the board serial number back. Flashing a Cube Orange with PX4 seems to only expose one port but I figure it was still worth noting since at some point this may become important for PX4 as well.

I'm not sure if this is OSX only but there is where I noticed it

julianoes, 2023-04-03:
@DonLakeFlyer so you're saying that it would be better if the serial number was exposed so that multiple flight controllers could be distinguished, right?

@davids5 what's your take on that?

davids5, 2023-04-03:
Part of this was done for the non intree BL. https://github.com/PX4/PX4-Bootloader/pull/41

julianoes, 2023-04-03:
@DonLakeFlyer ok, I understand that part. That's just because ArduPilot has a second port for debugging or SLCAN or other functionality.