PX4/PX4-Autopilot #15744: incremental build failure due to ROMFS processing

dagar, 2020-09-15:
With current master simply build (successfully), then attempt to build again (with or without modifications).

``` Console

dagar@dagar-desktop:~/git/Firmware$ make px4_fmu-v5_default 
[1/15] Performing build step for 'px4io_firmware'
ninja: no work to do.
[4/15] ROMFS: copying, generating airframes
FAILED: cd /home/dagar/git/Firmware/build/px4_fmu-v5_default/ROMFS && /usr/bin/python3 /home/dagar/git/Firmware/Tools/px_process_airframes.py --airframes-path /home/dagar/git/Firmware/build/px4_fmu-v5_default/etc/init.d --start-script /home/dagar/git/Firmware/build/px4_fmu-v5_default/etc/init.d/rc.autostart --board px4_fmu-v5 && /usr/bin/python3 /home/dagar/git/Firmware/Tools/serial/generate_config.py --rc-dir /home/dagar/git/Firmware/build/px4_fmu-v5_default/etc/init.d --serial-ports GPS1:/dev/ttyS0 TEL1:/dev/ttyS1 TEL2:/dev/ttyS2 TEL4:/dev/ttyS3 --config-files /home/dagar/git/Firmware/src/lib/battery/module.yaml /home/dagar/git/Firmware/src/drivers/distance_sensor/cm8jl65/module.yaml /home/dagar/git/Firmware/src/drivers/distance_sensor/leddar_one/module.yaml /home/dagar/git/Firmware/src/drivers/distance_sensor/sf0x/module.yaml /home/dagar/git/Firmware/src/drivers/distance_sensor/tfmini/module.yaml /home/dagar/git/Firmware/src/drivers/distance_sensor/ulanding_radar/module.yaml /home/dagar/git/Firmware/src/drivers/dshot/module.yaml /home/dagar/git/Firmware/src/drivers/gps/module.yaml /home/dagar/git/Firmware/src/drivers/optical_flow/thoneflow/module.yaml /home/dagar/git/Firmware/src/drivers/rc_input/module.yaml /home/dagar/git/Firmware/src/drivers/roboclaw/module.yaml /home/dagar/git/Firmware/src/drivers/telemetry/frsky_telemetry/module.yaml /home/dagar/git/Firmware/src/drivers/telemetry/hott/hott_telemetry/module.yaml /home/dagar/git/Firmware/src/drivers/telemetry/iridiumsbd/module.yaml /home/dagar/git/Firmware/src/modules/battery_status/module.yaml /home/dagar/git/Firmware/src/modules/mavlink/module.yaml && /home/dagar/.local/lib/python3.6/site-packages/cmake/data/bin/cmake -E touch romfs_copy.stamp
Aborting due to missing @type tag in file: '/home/dagar/git/Firmware/build/px4_fmu-v5_default/etc/init.d/airframes/2507_cloudship'
[4/15] Completed 'px4io_firmware'
ninja: build stopped: subcommand failed.
Makefile:217: recipe for target 'px4_fmu-v5_default' failed
make: *** [px4_fmu-v5_default] Error 1
```

Broken in https://github.com/PX4/Firmware/pull/15714.

bkueng, 2020-09-16:
Hmm, I can't reproduce, for me the tar is always extracted before px_process_airframes.py runs.
Maybe https://github.com/PX4/Firmware/pull/15747 helps?

dagar, 2020-09-16:
Weird that you aren't getting the same result. I'm still on Ubuntu 18.04.

Fixed by https://github.com/PX4/Firmware/pull/15714.

closed 2020-09-16 by dagar