PX4/PX4-Autopilot #15756: Sitl Build fails

yashom7, 2020-09-17:
Did make px4_sitl jmavsim on MacOS  to test the toolchain installation resulting in the build failing with the following error

```

[1/478] Building CXX object src/lib/flight_tasks/tasks/Utility/CMakeFiles/FlightTaskUtility.dir/ManualSmoothingZ.cpp.o
FAILED: src/lib/flight_tasks/tasks/Utility/CMakeFiles/FlightTaskUtility.dir/ManualSmoothingZ.cpp.o
/usr/local/bin/ccache /Library/Developer/CommandLineTools/usr/bin/c++ -DCONFIG_ARCH_BOARD_PX4_SITL -DENABLE_LOCKSTEP_SCHEDULER -DMODULE_NAME=\"FlightTaskUtility\" -D__CUSTOM_FILE_IO__ -D__PX4_DARWIN -D__PX4_POSIX -D__STDC_FORMAT_MACROS -Dnoreturn_function="__attribute__((noreturn))" -I../../boards/px4/sitl/src -I../../platforms/posix/src/px4/common/include -I. -Isrc/lib -I../../platforms/posix/src/px4/generic/generic/include -I../../platforms/common/include -I../../src -I../../src/include -I../../src/lib -I../../src/lib/matrix -I../../src/modules -I../../platforms/posix/include -Iexternal/Install/include -I../../src/lib/flight_tasks/tasks/Utility -I../../src/lib/flight_tasks/tasks/FlightTask -O2 -g -DNDEBUG -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk -fPIC -g -fdata-sections -ffunction-sections -fomit-frame-pointer -fmerge-all-constants -fno-signed-zeros -fno-trapping-math -freciprocal-math -fno-math-errno -fno-strict-aliasing -fvisibility=hidden -include visibility.h -Wall -Wextra -Werror -Warray-bounds -Wcast-align -Wdisabled-optimization -Wdouble-promotion -Wfatal-errors -Wfloat-equal -Wformat-security -Winit-self -Wlogical-op -Wpointer-arith -Wshadow -Wuninitialized -Wunknown-pragmas -Wunused-variable -Wno-missing-field-initializers -Wno-missing-include-dirs -Wno-unused-parameter -fcolor-diagnostics -fdiagnostics-absolute-paths -Qunused-arguments -Wno-unknown-warning-option -Wno-unused-const-variable -Wno-varargs -fno-exceptions -fno-rtti -fno-threadsafe-statics -Wreorder -Wno-overloaded-virtual -Wno-cast-align -std=gnu++14 -MD -MT src/lib/flight_tasks/tasks/Utility/CMakeFiles/FlightTaskUtility.dir/ManualSmoothingZ.cpp.o -MF src/lib/flight_tasks/tasks/Utility/CMakeFiles/FlightTaskUtility.dir/ManualSmoothingZ.cpp.o.d -o src/lib/flight_tasks/tasks/Utility/CMakeFiles/FlightTaskUtility.dir/ManualSmoothingZ.cpp.o -c ../../src/lib/flight_tasks/tasks/Utility/ManualSmoothingZ.cpp
In file included from ../../src/lib/flight_tasks/tasks/Utility/ManualSmoothingZ.cpp:38:
In file included from ../../src/lib/flight_tasks/tasks/Utility/ManualSmoothingZ.hpp:43:
/Users/yashomdighe/PX4/Firmware/platforms/common/include/px4_platform_common/module_params.h:81:20: fatal error: loop variable 'child' is always a copy because the range of type 'List<ModuleParams *>' does not return a reference [-Wrange-loop-analysis]
                for (const auto &child : _children) {
                                 ^
/Users/yashomdighe/PX4/Firmware/platforms/common/include/px4_platform_common/module_params.h:81:8: note: use non-reference type 'ModuleParams *'
                for (const auto &child : _children) {
                     ^~~~~~~~~~~~~~~~~~~
1 error generated.
[2/478] Building CXX object src/lib/flight_tasks/tasks/Utility/CMakeFiles/FlightTaskUtility.dir/ManualSmoothingXY.cpp.o
FAILED: src/lib/flight_tasks/tasks/Utility/CMakeFiles/FlightTaskUtility.dir/ManualSmoothingXY.cpp.o
/usr/local/bin/ccache /Library/Developer/CommandLineTools/usr/bin/c++ -DCONFIG_ARCH_BOARD_PX4_SITL -DENABLE_LOCKSTEP_SCHEDULER -DMODULE_NAME=\"FlightTaskUtility\" -D__CUSTOM_FILE_IO__ -D__PX4_DARWIN -D__PX4_POSIX -D__STDC_FORMAT_MACROS -Dnoreturn_function="__attribute__((noreturn))" -I../../boards/px4/sitl/src -I../../platforms/posix/src/px4/common/include -I. -Isrc/lib -I../../platforms/posix/src/px4/generic/generic/include -I../../platforms/common/include -I../../src -I../../src/include -I../../src/lib -I../../src/lib/matrix -I../../src/modules -I../../platforms/posix/include -Iexternal/Install/include -I../../src/lib/flight_tasks/tasks/Utility -I../../src/lib/flight_tasks/tasks/FlightTask -O2 -g -DNDEBUG -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk -fPIC -g -fdata-sections -ffunction-sections -fomit-frame-pointer -fmerge-all-constants -fno-signed-zeros -fno-trapping-math -freciprocal-math -fno-math-errno -fno-strict-aliasing -fvisibility=hidden -include visibility.h -Wall -Wextra -Werror -Warray-bounds -Wcast-align -Wdisabled-optimization -Wdouble-promotion -Wfatal-errors -Wfloat-equal -Wformat-security -Winit-self -Wlogical-op -Wpointer-arith -Wshadow -Wuninitialized -Wunknown-pragmas -Wunused-variable -Wno-missing-field-initializers -Wno-missing-include-dirs -Wno-unused-parameter -fcolor-diagnostics -fdiagnostics-absolute-paths -Qunused-arguments -Wno-unknown-warning-option -Wno-unused-const-variable -Wno-varargs -fno-exceptions -fno-rtti -fno-threadsafe-statics -Wreorder -Wno-overloaded-virtual -Wno-cast-align -std=gnu++14 -MD -MT src/lib/flight_tasks/tasks/Utility/CMakeFiles/FlightTaskUtility.dir/ManualSmoothingXY.cpp.o -MF src/lib/flight_tasks/tasks/Utility/CMakeFiles/FlightTaskUtility.dir/ManualSmoothingXY.cpp.o.d -o src/lib/flight_tasks/tasks/Utility/CMakeFiles/FlightTaskUtility.dir/ManualSmoothingXY.cpp.o -c ../../src/lib/flight_tasks/tasks/Utility/ManualSmoothingXY.cpp
In file included from ../../src/lib/flight_tasks/tasks/Utility/ManualSmoothingXY.cpp:38:
In file included from ../../src/lib/flight_tasks/tasks/Utility/ManualSmoothingXY.hpp:50:
/Users/yashomdighe/PX4/Firmware/platforms/common/include/px4_platform_common/module_params.h:81:20: fatal error: loop variable 'child' is always a copy because the range of type 'List<ModuleParams *>' does not return a reference [-Wrange-loop-analysis]
                for (const auto &child : _children) {
                                 ^
/Users/yashomdighe/PX4/Firmware/platforms/common/include/px4_platform_common/module_params.h:81:8: note: use non-reference type 'ModuleParams *'
                for (const auto &child : _children) {
                     ^~~~~~~~~~~~~~~~~~~
1 error generated.
[6/478] Building CXX object src/lib/flight_tasks/tasks/FlightTask/CMakeFiles/FlightTask.dir/FlightTask.cpp.o
FAILED: src/lib/flight_tasks/tasks/FlightTask/CMakeFiles/FlightTask.dir/FlightTask.cpp.o
/usr/local/bin/ccache /Library/Developer/CommandLineTools/usr/bin/c++ -DCONFIG_ARCH_BOARD_PX4_SITL -DENABLE_LOCKSTEP_SCHEDULER -DMODULE_NAME=\"FlightTask\" -D__CUSTOM_FILE_IO__ -D__PX4_DARWIN -D__PX4_POSIX -D__STDC_FORMAT_MACROS -Dnoreturn_function="__attribute__((noreturn))" -I../../boards/px4/sitl/src -I../../platforms/posix/src/px4/common/include -I. -Isrc/lib -I../../platforms/posix/src/px4/generic/generic/include -I../../platforms/common/include -I../../src -I../../src/include -I../../src/lib -I../../src/lib/matrix -I../../src/modules -I../../platforms/posix/include -Iexternal/Install/include -I../../src/lib/flight_tasks/tasks/FlightTask -O2 -g -DNDEBUG -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk -fPIC -g -fdata-sections -ffunction-sections -fomit-frame-pointer -fmerge-all-constants -fno-signed-zeros -fno-trapping-math -freciprocal-math -fno-math-errno -fno-strict-aliasing -fvisibility=hidden -include visibility.h -Wall -Wextra -Werror -Warray-bounds -Wcast-align -Wdisabled-optimization -Wdouble-promotion -Wfatal-errors -Wfloat-equal -Wformat-security -Winit-self -Wlogical-op -Wpointer-arith -Wshadow -Wuninitialized -Wunknown-pragmas -Wunused-variable -Wno-missing-field-initializers -Wno-missing-include-dirs -Wno-unused-parameter -fcolor-diagnostics -fdiagnostics-absolute-paths -Qunused-arguments -Wno-unknown-warning-option -Wno-unused-const-variable -Wno-varargs -fno-exceptions -fno-rtti -fno-threadsafe-statics -Wreorder -Wno-overloaded-virtual -Wno-cast-align -std=gnu++14 -MD -MT src/lib/flight_tasks/tasks/FlightTask/CMakeFiles/FlightTask.dir/FlightTask.cpp.o -MF src/lib/flight_tasks/tasks/FlightTask/CMakeFiles/FlightTask.dir/FlightTask.cpp.o.d -o src/lib/flight_tasks/tasks/FlightTask/CMakeFiles/FlightTask.dir/FlightTask.cpp.o -c ../../src/lib/flight_tasks/tasks/FlightTask/FlightTask.cpp
In file included from ../../src/lib/flight_tasks/tasks/FlightTask/FlightTask.cpp:1:
In file included from ../../src/lib/flight_tasks/tasks/FlightTask/FlightTask.hpp:44:
/Users/yashomdighe/PX4/Firmware/platforms/common/include/px4_platform_common/module_params.h:81:20: fatal error: loop variable 'child' is always a copy because the range of type 'List<ModuleParams *>' does not return a reference [-Wrange-loop-analysis]
[... cut ...]
/Users/yashomdighe/PX4/Firmware/platforms/common/include/px4_platform_common/module_params.h:81:8: note: use non-reference type 'ModuleParams *'
                for (const auto &child : _children) {
                     ^~~~~~~~~~~~~~~~~~~
1 error generated.
ninja: build stopped: subcommand failed.
make: *** [px4_sitl] Error 1
```

Clang version is :
Apple clang version 12.0.0 (clang-1200.0.31.1)
Target: x86_64-apple-darwin19.6.0
Thread model: posix
  
The build succeeds for other targets. I have tested it for fmu-v5 and fmu-v4

iannase, 2020-09-19:
Seems like a problem when updating to xCode 12 on MacOS with clang version 12.0.0

Here's how I disabled that warning:
`cd Firmware
grep -rl '\-Wall' ./ | xargs sed -i '' 's/\-Wall/\-Wno\-range\-loop\-analysis/g'`

-Wno-range-loop-analysis disables that clang flag in all the cmake files.

yashom7, 2020-09-19:
That Solved it. Thanks. 
Any idea why it throws that error?

bkueng, 2020-09-21:
Let's fix it for everyone. @yashom7 @iannase can you test #15770?

yashom7, 2020-09-21:
Yep sure. I will try that

yashom7, 2020-09-21:
@bkueng I made similar changes to #15770 in https://github.com/PX4/Firmware/blob/4d977727560507473fd9db2b471d5d08e900729d/src/include/containers/IntrusiveSortedList.hpp#L136 
 
The build succeeds after this but I am getting some java errors on jmavsim. I will make sure it is not related and make a pull request

PR #15780 (2020-09-21): fix 15756 so clang 12 doesn't complain
fixes #15756 by adding to #15770 
and making similar changes in IntrusiveSortedList
closed 2020-09-21 by dagar