PX4/PX4-Autopilot #15759: Build Procedure on Mac OS now stops with errors (all targets)

taileron, 2020-09-18:
**Describe the bug**
Build Procedure on Mac OS now stops with errors (all targets)

**To Reproduce**
clone the repository to e.g. firmware and make omnibus_f4sd_default:

[133/751] Generating apps/libapps.a, apps/platform/.built
FAILED: NuttX/apps/libapps.a NuttX/apps/platform/.built 
cd /Users/ahuber/src/Firmware/build/omnibus_f4sd_default/NuttX/apps && find /Users/ahuber/src/Firmware/build/omnibus_f4sd_default/NuttX/apps -name *.o -delete && make --quiet --no-print-directory TOPDIR="/Users/ahuber/src/Firmware/build/omnibus_f4sd_default/NuttX/nuttx"
make[2]: flock: No such file or directory
make[2]: *** [.built] Error 1
make[1]: *** [/Users/ahuber/src/Firmware/build/omnibus_f4sd_default/NuttX/apps/builtin_all] Error 2
[138/751] Generating nuttx/sched/libsched.a
ninja: build stopped: subcommand failed.
make: *** [omnibus_f4sd_default] Error 1

**Expected behavior**
before this procedure worked flawlessly

TSC21, 2020-09-18:
Please update your `px4-dev` brew install. `flock` is already [included](https://github.com/PX4/homebrew-px4/commit/ee084686eac90d80ab8a52bd345cad814cfd952e) on the latest recipe.

closed 2020-09-18 by TSC21

taileron, 2020-09-18:
Thanks a lot, after a long lasting BREW update procedure, everything works again as desired !