PX4/PX4-Autopilot #15769: Qgc daily build version does not display uavcan battery voltage

cuhome, 2020-09-21:
X7 series flight controllers cannot use QGC 4.0.10 to update the firmware online, so I am using the latest daily build qgc to replace it. But I found that the QGC built daily has updated the way to get the battery status. When I use uavcan batteries, qgc does not display voltage information. It is a malfunction of the firmware and qgc collaboration, so I post it here.
Related discussion：
https://github.com/mavlink/qgroundcontrol/issues/9079

cuhome, 2020-09-22:
@dagar @davids5 How can we make the firmware follow the mavlink specification?

cuhome, 2020-09-22:
[image]
We think that filling the first cell with full voltage is a quick solution, but it does not seem to meet the requirements. The difficulty lies in how the flight controller knows that the battery consists of several cells.

saengphet, 2020-09-25:
Hello, @cuhome  Have you try to use CUAV HV PM + X7pro + V5+ carrier board? I tried to use that config. and after I connected the battery into CUAV HV PM that hooks up to POWER 1, no battery voltage appeared on QGC. I checked msg "battery_status" and found that FCU was able to read the voltage correctly, whoever battery_status.connected was set to False instead of True. For this reason, the QGC will not show voltage and accumulated consumption and I'm struggling to solve this problem.

cuhome, 2020-10-11:
> Hello, @cuhome Have you tried to use CUAV HV PM + X7pro + V5+ carrier board? I tried to use that set and after I connected the battery into CUAV HV PM which hooks up to POWER 1 of 5+ carrier board, no battery voltage appeared on QGC (battery icon is N/A). I checked msg "battery_status" and found that FCU was able to read the voltage correctly (battery_status.voltage_v), however, "battery_status.connected was False instead of True". Therefore, the QGC will not show voltage and accumulated consumption and I'm struggling to solve this problem.
> 
> For CANPMU, related problem? #15306 , I temporarily fixed battery remaining problem by applying BAT2 parameters into uavcan battery module and it works fine QGC was received all battery status from CANPMU.

I have checked it according to your description. The base board of v5+ and x7 is different in power supply design. We have found this problem. If you need to install x7 core to v5+ base and use analog power, you need to modify the code.

cuhome, 2020-10-11:
[image]

You need to follow the arrows to switch them.

closed 2026-04-10 by github-actions[bot]