PX4/PX4-Autopilot #15775: UAV kept following mission even RC was lost, ecpected RTL

maxhofidrone, 2020-09-21:
**Describe the bug**
I uploaded a mission with QGC and started it successfully with QGC. Mid-Mission I turned of the remote RC to trigger the RC Loss failsafe option, but the UAV kept flying the mission. QGC warned me of the lost RC connection and also informed me about the regained connection after I turned back on the RC controller.

**To Reproduce**
see above

**Expected behavior**
I expected the UAV to return to launch after an RC loss of 0.5 seconds as it is set up in the safty parameters of  QGC

**Log Files and Screenshots**

-  flight log  https://review.px4.io/plot_app?log=9eb96a04-b6e9-4918-9968-cdf94234e375

Safety settings in QGC:
[image]

Exported parameters for QGC (needed to change the data type to .txt to upload to github)
[parameters.txt](https://github.com/PX4/Firmware/files/5255463/parameters.txt)

**Drone (please complete the following information):**
- Holybro X500 https://docs.px4.io/master/en/frames_multicopter/holybro_x500_pixhawk4.html
- RC-remote: Futaba T8J

dagar, 2020-09-21:
This is currently how it's intended to work because 1) AUTO modes don't require RC and 2) you still had contact with the vehicle through the data link.

FYI @julianoes

ryanjAA, 2020-09-21:
That’s actually a little misleading come to think of it. If failsafe is set to return after 0.5s of losing RC, it probably should return independent of mode or at least there should be another option (when in mission/when not), etc. At the very least this should be in the docs but it should probably be setup to return as default if RC is lost even if in mission (with a parameter to alter that). Current docs don’t say that, just that if RC is lost the result is X under RC failsafe. @hamishwillee

dagar, 2020-09-21:
> That’s actually a little misleading come to think of it.

I agree it's misleading, but I think the implementation makes sense. I'm not sure how we make these things clear from a QGC perspective. The "dream" (from my perspective at least) is that we get to the point where we can generate useful documentation from the actual state machine (transitions for failsafes, mode requirements, etc).

ryanjAA, 2020-09-21:
That would be good so the codebase dictates what’s in the docs. No ambiguity would be great. 

Ok, so the question then is how or what setting can be set to RTL when in mission mode when the RC link drops? It sounds like that isn’t possible.

ryanjAA, 2020-09-22:
I still think failsafe from RC should be based on if we have RC. Not the mode we are in. That could be a parameter to toggle it but if you can RTL at anytime from the RC (ie set return to a channel), the RC link then becomes the most important link and typically the furthest reaching one. At the same time having the ability to disregard the RC state and conduct a mission is equally important.

dagar, 2020-09-22:
If the data link loss is configured and you have a data link then you don't have to have RC in a mode that doesn't use it for anything. We could add a mechanism to override that if you think it's important.

ryanjAA, 2020-09-22:
Ya, I think it’s important to have the option. I’m just thinking that segregates links give more predicable and separate control from a safety perspective. I prefer an RC link with a slow data link over that as a backdoor and a dedicated datalink. Either SIK based or high speed. 2 physical links but 3 ways in. 

9/10 times the telemetry will die first given it’s a higher Frequency and if it’s high speed almost always first so I don’t think killing a mission is a good idea based on that but the option should be there (which I believe it is).  

Then if all that’s left is the RC and the datalink is gone, that is used to send the plane home if need be (return mode or what not) and the backdoor telemetry is if you know you’re past your high speed link range and you do need to get in but let’s forget that exists for a moment (that’s more of a personal setup having the third link).

 I just think it makes sense that if RC failsafe is set to do something, it should do it in a higher level than the mode we are in. 

You also run into issues with certain high speed links like Microhard units. That’s not correctly plumbed into qgc yet so if that’s your datalink you have no failsafe (other than battery). I definitely think RC failsafe makes sense to be mode independent (with a parameter to toggle what you’d like). Seems safest too.

maxhofidrone, 2020-09-22:
Dear @dagar and @ryanjAA ,
thanks a lot for the answers and insights of the failsafe options. I have the same opinion as @ryanjAA - in case I loose my RC connection during the mission flight I want the failsafe action triggered. The reason why I have a RC in mission mode is, that I want to terminate the mission flight as fast as possible (=switching modes) in case something goes wrong.  Loosing the RC, switching to the Laptop with QGC to terminate the mission might cost some important seconds. But I'm aware of that now and can include QGC more in my mission supervision. 

From my side this issue is closed now. But I can leave it open if needed for the documentation? 

Best regards!
Max

jimdgit, 2020-09-29:
Often many do NOT want fail safe in a mission - they want the mission to continue.
For some that is the whole point of on board waypoints. mission can continue even if RC is lost.
Need to be able to configure this,

hamishwillee, 2020-09-29:
@dagar The most obvious and flexible solution for a user is a second auto-mode-specific failsafe action that defaults to "no action". The failsafe for manual or auto modes are then then always obeyed if set, but might have no action.

@ryanjAA FYI I have updated PX4 safety docs to clarify that the failsafe only applies to manual modes:  https://github.com/PX4/px4_user_guide/pull/817 . Feel free to create your own PR updates if you see things like this :-)

I also have raised issue on QGC - they will have to work out how to fix it - presumably some note: https://github.com/mavlink/qgroundcontrol/issues/9092

FWIW 
- Recently there was a change that stick movement in automatic modes switches from auto to position mode *by default*. To me that implies that there is an expectation that RC is the important default fallback, which in turn implies to me that people can reasonably expect that RC loss will result in a safety action.

jimdgit, 2020-09-30:
It should be a checkbox in the Mission Start dialog  - Failsafe Action/Continue mission on Failsafe.
Just that simple and easy,
This makes it more accessible, easier to find and more understandable.
It's right there as part of the mission.
It's not a mode, it's just a mission option.

dagar, 2020-09-30:
A relatively easy immediate solution would be to add another commander parameter to configure this behavior (auto/mission requirements).

jimdgit, 2020-09-30:
> A relatively easy immediate solution would be to add another commander parameter to configure this behavior (auto/mission requirements).

Perfect.

hamishwillee, 2025-03-27:
Removing documentation tag. No other docs work for this in the current state.