PX4/PX4-Autopilot #15788: nxp_fmuk66-v3_default intermittant test hardfault

dagar, 2020-09-22:
Noticed on the test rack. http://ci.px4.io:8080/blue/organizations/jenkins/PX4_misc%2FFirmware-hardware/detail/pr-rotation_sync/4/pipeline/1808

[image]

dagar, 2020-09-22:
FYI @davids5

davids5, 2020-10-13:
@dagar Has this happened again since https://github.com/PX4/Firmware/pull/15948 came in? Is there a way to get notifications from the test rack on classes of failure?

dagar, 2020-10-13:
> @dagar Has this happened again since #15948 came in? Is there a way to get notifications from the test rack on classes of failure?

You can see it per branch. http://ci.px4.io:8080/blue/organizations/jenkins/PX4_misc%2FFirmware-hardware/activity?branch=master

Right now things are in a bad shape with a number of boards getting stuck in the hard fault loop. http://ci.px4.io:8080/blue/organizations/jenkins/PX4_misc%2FFirmware-hardware/detail/master/2397/pipeline/1410

davids5, 2020-10-13:
> You can see it per branch. http://ci.px4.io:8080/blue/organizations/jenkins/PX4_misc%2FFirmware-hardware/activity?branch=master
>
> Right now things are in a bad shape with a number of boards getting [...]

This should https://github.com/PX4/Firmware/pull/15957 fix all that