PX4/PX4-Autopilot #15790: Unstable flight with multicopter in position control with v1.11

NAmmann, 2020-09-22:
**In position control the angles 'oscilate' resulting in an unstable flight**
After updating the Pixhawk 4 to v1.11 the first flight test resulted in an unstable flight performance resulting from oscilation of the angles during manual controlled flight in position control mode. This behaviour was not visible in manual mode, mission mode nor RTL mode. Resetting position/velocity control parameters to default did not change that behaviour. A downgrade to v1.10 fixed that behaviour immediately. 

**To Reproduce**
Steps to reproduce the behavior:
1. Upgrade Pixhawk 4 to V1.11 with parameters as specified in the logs.
2. Takoeoff in automatic or manual flight mode
2. Switch to position control
3. Do a manual controlled flight
4. See logs and observe flight performance

**Expected behavior**
Smooth flight trajectory without oscilation of the angles.

**Log Files and Screenshots**
Here are the links to the flight tests: 
- V1.11 with working parameters from v1.10 (Manual, Hold, Position, RTL, Manual): https://logs.px4.io/plot_app?log=0a8a6bbe-a49e-4b01-ba44-9b1c0261f0c3
- V1.11 with working parameters from v1.10 (Mission): https://logs.px4.io/plot_app?log=a9898b24-41e1-4f0e-89ea-95d40edb1dab
- V1.11 with default parameters for pos ctrl (Manual, Position, Manual): https://logs.px4.io/plot_app?log=6ee87e24-ed32-4702-8939-28c1d86f5a8d
- V1.10 downgrad (manual, Position, Manual): https://logs.px4.io/plot_app?log=de7ca3fd-16fe-419d-b633-a469fa96fa75

And some screenshots to describe the differences:
- Pitch angle for manual flight in position control with v1.11: [image]
- Pitch angle for automatic flight in mission mode [image]
 with v1.11:
- Pitch angle for manual flight in position control with v1.10: [image]

**Hexacopter with Pixhawk 4**
- 11kg Hexacopter based on DJI M600

dagar, 2020-09-22:
[image]

As an aside I'd suggest decreasing IMU_GYRO_CUTOFF to 30 Hz.

MaEtUgR, 2020-09-22:
> This behaviour was not visible in manual mode, mission mode nor RTL mode.

That's a very important clue. It tells us that the attitude setpoint oscillations are very likely excited by the manual joystick input setpoint generation and less likely come from the position control itself.

jimdgit, 2020-09-22:
FWIW It's unstable in javasim s well.
Just did a totally clean build of release/1.11 branch and in javasim it starts out ok, but the goes unstable.
Fly in reverse and it does not brake well and then just bounces around,

NAmmann, 2020-09-22:
> > This behaviour was not visible in manual mode, mission mode nor RTL mode.
> 
> That's a very important clue. It tells us that the attitude setpoint oscillations are very likely excited by the manual joystick input setpoint generation and less likely come from the position control itself.

But if you look to the RC inputs there are no such inputs. Also this behavior is not present with version v1.10. Therefore I do not think that this is based on the input.

bresch, 2020-09-23:
In 1.10 we had a low-pass filter on the RC signals that we removed in https://github.com/PX4/Firmware/pull/14896/commits/f9f62e83965917b311bc079f6c5d5ac4a49fe7bc
This could explain the change in behavior.
FYI: @MaEtUgR

MaEtUgR, 2020-09-23:
I had a closer look at the log and the input (at least in the log frequency) seems fine. Not having the issue when flying manual/stabilized supports that assumption since by default (and in the log) additional input filtering `MC_MAN_TILT_TAU` is disabled there. It looks more like this nervous attitude setpoint changes come in at either velocity control or acceleration feed-forward level. It's really hard to see since only the attitude setpoint is logged at the full 50Hz update rate. `local_position_setpoint` only at 10Hz, `trajectory_setpoint` even just at 5Hz.
[image]

Maybe we can capture the problem in jmavsim simulation when logging full rate like @jimdgit started trying.

AlexKlimaj, 2021-03-16:
@NAmmann seeing this on a Cube black updating from 1.10 to 1.11 as well. Did you manage to solve it?

NAmmann, 2021-03-18:
The problem is that parameters change their default values. See MPC_JERK_MAX for example. It changed from 20 to 8. Maybe the internal processing of these parameters changed as well. 
Resetting the MPC_XXX parameters back to the new default values solved the issue for us.