PX4/PX4-Autopilot #15795: Regression with source command

julianoes, 2020-09-23:
The recent change (https://github.com/PX4/Firmware/commit/6e1078c96ea22f0a8c64dd6a4956661a086644db) using `source` instead of `.` doesn't work on my Arch Linux:

```
px4 starting.

INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
/home/julianoes/src/Firmware/build/px4_sitl_default/bin/px4-alias.sh: line 42: `source': is a special builtin
ERROR [px4] Startup script returned with return value: 512
```

FYI @dagar @davids5

MaEtUgR, 2020-09-23:
I have the same with `GNU bash, version 5.0.17`. I expected other platforms than Cygwin having these problems since Cygwin doesn't reinvent bash but rather just keeps it up to date. The last discussion about this topic was: https://github.com/PX4/Firmware/pull/15139#discussion_r486950808

It breaks with https://github.com/PX4/Firmware/commit/185ffe9e13d3b661a4210041cce5bfd22ee1d695 and gives me
```
/home/maetugr/Firmware/build/px4_sitl_default/tmp/rootfs/etc/init.d/rc.vehicle_setup: line 81: /etc/init.d/rc.interface: No such file or directory
ERROR [px4] Startup script returned with return value: 256
```

With https://github.com/PX4/Firmware/commit/6e1078c96ea22f0a8c64dd6a4956661a086644db the error changes to
```
/home/maetugr/Firmware/build/px4_sitl_default/bin/px4-alias.sh: line 42: `source': is a special builtin
ERROR [px4] Startup script returned with return value: 512
```

I think we should listen to @bkueng : https://github.com/PX4/Firmware/pull/15139#discussion_r487719343

> The main difficulty is that both source and . require an absolute path.
If we need to avoid overriding altogether, we can add a prefix variable, $PWD on POSIX, and / on NuttX: . ${R}etc/init.d/rc.fw_defaults

dagar, 2020-09-23:
I'll dig into this later if no one has an immediate solution on hand.

If possible we should also probably favor `.` over `source` to maximize compatibility with POSIX sh.

coderkalyan, 2020-09-23:
Able to replicate this on Arch as well. Seems to be bash-specific, since Arch and Cygwin symlink bash to /bin/sh while Ubuntu uses dash (which is technically more POSIX compliant). Tested by relinking /bin/sh to dash and it works fine.

MaEtUgR, 2020-09-26:
I did some more tests and replacing `source` with e.g. `so` in https://github.com/PX4/Firmware/blob/a24488328f175e4657ad449996c42260e4da9c92/platforms/posix/src/px4/common/px4-alias.sh_in#L31 gets rid of the error 512 ``source': is a special builtin` but then it fails because of `rc.vehicle_setup` invoking NuttX path scripts e.g. with multicopter here: https://github.com/PX4/Firmware/blob/a24488328f175e4657ad449996c42260e4da9c92/ROMFS/px4fmu_common/init.d/rc.vehicle_setup#L80-L84

After reverting those to `sh` instead of `source` it works again. So I guess a variable for the path of invoked scripts which is empty on NuttX would make sense right?

MaEtUgR, 2020-10-20:
This was resolved by #15970

closed 2020-10-20 by MaEtUgR