PX4/PX4-Autopilot #15801: X Quad: CubeOrange / DShot AUX-4 output not mapped correctly

ecmnet, 2020-09-23:
**Describe the bug**
Using 1.11 release, I have a strange behavior with DShot and CubeOrange on a X-Quad. When armed Motor 4 is spinning at 50%. Other Motors/Outputs work correctly. Using QGC motor-test, Motor 4 is not controllable at all (does not spin). With NSH motor_test, (motor_test test -m 4 -p 20) the motor does not start as well. However using motor_test test **-m 5** -p 20, the motor spins and seems to be controllable. Motors are connected to AUX in DShot mode as described in the documentation with IO = 0 set. Motor 4 is connected to AUX-4. Seems to be a kind of mapping issue between AUX4 and AUX5 on CubeOrange.

dagar, 2020-09-23:
We need to verify the timers and the dshot motor assignment.

https://github.com/PX4/Firmware/blob/0a607bdc676a62a3839a833897e983e9dc944d29/boards/cubepilot/cubeorange/src/timer_config.cpp#L41-L48

https://github.com/PX4/Firmware/blob/0a607bdc676a62a3839a833897e983e9dc944d29/boards/cubepilot/cubeorange/src/board_config.h#L148

dagar, 2020-09-23:
Can you comment on anything else? Have you done anything that would verify the mapping of AUX1,2,3? Have you tried PWM output?

dagar, 2020-09-23:
FYI @mirkix @JacobCrabill

dagar, 2020-09-23:
Ha, looks like inherited copypasta from the Holybro Durandal, the first STM32H7 board we supported.

ecmnet, 2020-09-23:
PWM output works fine. No issues here. Think it is related to https://github.com/PX4/Firmware/blob/0a607bdc676a62a3839a833897e983e9dc944d29/boards/cubepilot/cubeorange/src/board_config.h#L148

dagar, 2020-09-23:
> PWM output works fine. No issues here. Think it is related to

Yes I think so as well, can you give https://github.com/PX4/Firmware/pull/15802 a try?

ecmnet, 2020-09-24:
I can confirm that the fix works for me. Perfect!

dagar, 2020-09-24:
Fixed by https://github.com/PX4/Firmware/pull/15802. Thanks for the quick feedback @ecmnet.

closed 2020-09-24 by dagar