PX4/PX4-Autopilot #15810: hovergames disarm in air

dagar, 2020-09-24:
https://review.px4.io/plot_app?log=37ee74be-7c86-41b9-b172-115dcd91cd5b
https://www.youtube.com/watch?v=6e8kV0kW_Gk

[image]

@igalloway

dagar, 2020-09-24:
Disarmed by safety button while still in the air.

I would propose that we remove the ability for the safety button to disarm (at least by default).

landonh12, 2020-09-24:
Here is the flight log: https://review.px4.io/plot_app?log=37ee74be-7c86-41b9-b172-115dcd91cd5b

dagar, 2020-09-25:
Going by the details in your log the change in https://github.com/PX4/Firmware/pull/15811 should have prevented the disarm in air.

I'd still like to more explicitly have the safety disarm behavior entirely disabled by default, including px4io side.

dagar, 2020-12-27:
Fixed by https://github.com/PX4/PX4-Autopilot/pull/15811.

closed 2020-12-27 by dagar