PX4/PX4-Autopilot #15816: px4_ioctl(g_pwm_fd, PWM_SERVO_SET(0),output(0));can not work in v1.11 release

kcx064, 2020-09-25:
When I use **px4_ioctl** to control motors in v1.11 release, motors do not run anyway. Does the **px4_ioctl** have been cancelled in v1.11 release?

dagar, 2020-09-25:
Can you share any specifics? It's still used by vtol_att_control. https://github.com/PX4/Firmware/blob/60252dde08ac33a97bda3b8497035483b250097b/src/modules/vtol_att_control/vtol_type.cpp#L102

kcx064, 2020-09-26:
For Quadcopter, I use these code below to control motor in v1.11, but the PWM pulse width is always 900us.

` const char *g_pwm = PWM_OUTPUT0_DEVICE_PATH;`
`int ret= px4_open(g_pwm, 0);`

`//  No output (always 900) after PX4 armed.`
`px4_ioctl(ret , PWM_SERVO_SET(0), 1500);`
`px4_ioctl(ret , PWM_SERVO_SET(1), 1500);`
`px4_ioctl(ret , PWM_SERVO_SET(2), 1700);`
`px4_ioctl(ret , PWM_SERVO_SET(3), 1500);`

dagar, 2020-09-26:
Do you have a mixer loaded? Which board is this?

The output modules are really focused around subscribing to `actuator_controls` from the controllers, then mixing and writing outputs. The use case of directly setting PWM outside of that module gets in the way of that. If you have a mixer loaded and want to circumvent that you can use `test_motor` https://github.com/PX4/Firmware/blob/master/msg/test_motor.msg.

kcx064, 2020-12-04:
Thank you. 
In fact, I use simulink to generate code to realize my algorithm which need to directly set PWM. A mixer 
integrated with controller is dynamic. With PX4 **v1.10**,  some code generated by the simulink model file shown as below can be used to set PWM directly.
[image]

But with PX4 **v1.11.2** and **v1.11.1**, the same code generated by  the simulink file can not work as before.
Several board include Pixhawk, Pixhawk 2(Cube Black), Cube Orange have been used to run my custom code but they can not work well with PX4 **v1.11** to set PWM directly.

The code generated by Simulink are shown as below:

 ```c
/*
 * File: testpwm.c
 *
 * Code generated for Simulink model 'testpwm'.
 *
 * Model version                  : 1.25
 * Simulink Coder version         : 9.4 (R2020b) 29-Jul-2020
 * C/C++ source code generated on : Fri Dec  4 17:36:36 2020
 *
 * Target selection: ert.tlc
 * Embedded hardware selection: ARM Compatible->ARM Cortex
 * Code generation objectives: Unspecified
 * Validation result: Not run
 */

#include "testpwm.h"
#include "testpwm_private.h"

const char *g_pwm_device = PWM_OUTPUT0_DEVICE_PATH;
int g_pwm_fd = -1;
bool g_pwm_enabled = false;

/* Block signals (default storage) */
B_testpwm_T testpwm_B;

/* Block states (default storage) */
DW_testpwm_T testpwm_DW;

/* Real-time model */
static RT_MODEL_testpwm_T testpwm_M_;
RT_MODEL_testpwm_T *const testpwm_M = &testpwm_M_;

/* Model step function */
void testpwm_step(void)
{
  /* S-Function (sfun_px4_input_rc): '<Root>/input_rc'
   *
   * Block description for '<Root>/input_rc':
   *  RC Input Block
   *
   *  This block provides user input control to the model.
   *  It uses the input_rc uORB topic.
   *
   *  The user has the ability to choose which channels are available as outputs from this block and also some optional outputs. These include
   *  Channels 1 through 18
   *  double data type indicating the PWM value from the controller
   *  measured pulse widths for each of the supported channels
   *  Channel Count
   *  uint32 data type of the number of channels which are detector by the PX4
   *  RC Failsafe
   *  boolean data types indicating that the RC Tx is sending the FailSafe signal (if equipped and properly setup)
   *  explicit failsafe flag: true on TX failure or TX out of range , false otherwise.
   *  Only the true state is reliable, as there are some (PPM) receivers on the market going into failsafe without telling us explicitly.
   *  RC Input Source
   *  Enumeration data type indicating which source the RC input is from.
   *  Valid values are found in the ENUM file: RC_INPUT_SOURCE_ENUM.m
   *            Enumeration members for class 'RC_INPUT_SOURCE_ENUM':
   *            RCINPUT_SOURCE_UNKNOWN         (0)
   *            RCINPUT_SOURCE_PX4FMU_PPM      (1)
[... 276 lines cut ...]
   *  This is usually a function of the RC Tx in combination with other flight modes programmed in the Simulink model by the user.
   *  The block has 8 available ports (data type uint16) which can be selectively chosen.
   *  These correspond to the 8 PWM output ports on the px4fmu hardware.
   */

  /* disable pwm outputs */
  px4_ioctl(g_pwm_fd, PWM_SERVO_SET(0), PWM_MOTOR_OFF);
  px4_ioctl(g_pwm_fd, PWM_SERVO_SET(1), PWM_MOTOR_OFF);
  px4_ioctl(g_pwm_fd, PWM_SERVO_SET(2), PWM_MOTOR_OFF);
  px4_ioctl(g_pwm_fd, PWM_SERVO_SET(3), PWM_MOTOR_OFF);
  px4_ioctl(g_pwm_fd, PWM_SERVO_DISARM, 0);
  px4_ioctl(g_pwm_fd, PWM_SERVO_CLEAR_ARM_OK, 0);
  g_pwm_enabled = false;

  /* Close handle used in the PWM S-Function Block: <Root>/PWM_output */
  px4_close(g_pwm_fd);
}

/*
 * File trailer for generated code.
 *
 * [EOF]
 */

```

closed 2021-07-11 by kcx064