PX4/PX4-Autopilot #15822: Eclipse: Selected text cannot be mapped to a symbol name

srappensperger, 2020-09-26:
Hi guys,
I try to redesign some hardware functions.
As programming IDE I use **Eclipse, Version: 2020-06** (Ubuntu 20.04.1 LTS)

For the project setup I imported the px4-Firmware project and the settings via the file eclipse.project.
Everything works fine so far except for the **Eclipse function "Open Declaration"**.
When I try to call a variable declaration from a board.h file, it should normally refer to a file in the path "plattforms/nuttx/NuttX/nuttx/...", but instead of doing that the **error** "**Selected text cannot be mapped to a symbol name**" appears. 
If someone has an idea I will be very grateful.

dagar, 2020-09-26:
For some context the Eclipse project files were something I added a few years ago in the hope that we could get decent IDE behavior. I eventually gave up on Eclipse and switched to VSCode where we can get much better results due to directly integrating with the underlying build system (cmake).

If anyone is interested in maintaining the eclipse project file I'd be happy to keep it around, otherwise it might be better to delete them.

srappensperger, 2020-09-26:
Thanks a lot for your quick answer @dagar :smile: 
I followed your suggestion and tried VSCode. However after building the FMU-v5 board the "Go to Definitions" functionality still does not work (e.g.: GPIO_USART1_RX_2). Do you have any suggestions how to fix this?

dagar, 2020-09-26:
> Thanks a lot for your quick answer @dagar 
> I followed your suggestion and tried VSCode. However after building the FMU-v5 board the "Go to Definitions" functionality still does not work (e.g.: GPIO_USART1_RX_2). Do you have any suggestions how to fix this?

Did you select px4_fmu-v5 within vscode and let it configure + build? 

The other piece to be aware of is that the NuttX build is an external Makefile project only called by the PX4 cmake build, so properly resolving everything all the way into NuttX is a different story. For this and many other reasons I started a NuttX cmake port, but it's a surprisingly massive effort and the upstream project seemed opposed.

Hoeze, 2020-09-26:
Hi, I got the same problem:
Resolving definitions from NuttX does not work.
When working with plain NuttX makefile in Eclipse, I can include the preprocessor macro file in the config and Eclipse resolves all definitions.
With PX4, this does not work.

srappensperger, 2020-09-26:
> Did you select px4_fmu-v5 within vscode and let it configure + build?

Yes, I could configure and build px4_fmu-v5 without error messages.

> The other piece to be aware of is that the NuttX build is an external Makefile project only called by the PX4 cmake build, so properly resolving everything all the way into NuttX is a different [...]

Did you mean that this is also for you not possible?

dagar, 2020-09-26:
> Did you mean that this is also for you not possible?

Yes, currently within NuttX I revert to searching for symbols. You can get okay results if you enable the tag parser in VSCode, but it's a bit of a cpu hog, especially if you're constantly blowing away the build directory.

closed 2020-10-11 by srappensperger