PX4/PX4-Autopilot #15826: Holybro Kakute F7 Motors 0 and 1 are not working in 1.11.0

justas-, 2020-09-27:
**Describe the bug**
Motors 0 and 1 are not operational in 1.11 release of Px4.

**To Reproduce**
Steps to reproduce the behavior:

1. Install Firmware 1.10. Run "motor_test iterate." Observe all 4 motors spinning.
2. Install Firmware 1.11. Run "motor_test iterate." Observe motors 0 and 1 not spinning. 

Tested using PWM ESCs.

**Expected behavior**
All 4 motors working.

**Drone (please complete the following information):**
Holybro_Kakute_F7 using project provided builds.

**Additional context**
Add any other context about the problem here.

dagar, 2020-09-27:
Do the motors work at all? Are they spinning when you arm?

dagar, 2020-09-27:
@bkueng have you tried a Kakute recently?

justas-, 2020-09-27:
I can arm the drone and 2 motors are spinning. nsh log below:

sercon: Registering CDC/ACM serial driver
sercon: Successfully registered the CDC/ACM serial driver
HW arch: HOLYBRO_KAKUTEF7
FW git-hash: 71db0903a910d35a18f10d74c851446a6f66d61f
FW version: Release 1.11.0 (17498367)
OS: NuttX
OS version: Release 8.2.0 (134349055)
OS git-hash: ec20f2e6c5cc35b2b9bbe942dea55eabb81297b6
Build datetime: Sep  7 2020 20:27:55
Build uri: localhost
Toolchain: GNU GCC, 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]
PX4GUID: 000200000000343334343436510b0028002f
MCU: STM32F74xxx, rev. Z
Board defaults: /etc/init.d/rc.board_defaults
INFO  [dataman] Unknown restart, data manager file '/fs/microsd/dataman' size is 362560 bytes
ERROR [param] Parameter LIGHT_EN_BLINKM not found
Board sensors: /etc/init.d/rc.board_sensors
icm20689 #0 on SPI bus 4 (devid=0x3c)
bmp280 #0 on I2C bus 1 (external, equal to '-b 1')
hmc5883 #0 on I2C bus 1 (external, equal to '-b 1')
ERROR [param] Parameter TRIG_MODE not found
Board extras: /etc/init.d/rc.board_mavlink
INFO  [mavlink] mode: Config, data rate: 800000 B/s on /dev/ttyACM0 @ 57600B
INFO  [mavlink] mode: Onboard, data rate: 57600 B/s on /dev/ttyS1 @ 921600B
ERROR [mavlink] offboard mission init failed (-1)
INFO  [init] Mixer: /etc/mixers/quad_x.main.mix on /dev/pwm_output0
ERROR [param] Parameter TC_A_ENABLE not found
ERROR [param] Parameter TC_B_ENABLE not found
ERROR [param] Parameter TC_G_ENABLE not found
ERROR [param] Parameter MNT_MODE_IN not found
Board extras: /etc/init.d/rc.board_extras
INFO  [logger] logger started (mode=all)

NuttShell (NSH)
nsh> [KINFO  [logger] Start file log (type: full)
INFO  [logger] [logger] /fs/microsd/log/2020-09-27/16_00_13.ulg
INFO  [logger] Opened full log file: /fs/microsd/log/2020-09-27/16_00_13.ulg
INFO  [commander] Takeoff detected

bkueng, 2020-09-28:
I see the same - DShot works though.

bkueng, 2020-10-02:
@justas- can you test https://github.com/PX4/Firmware/pull/15867?

justas-, 2020-10-02:
Will do either today or over the weekend.

justas-, 2020-10-03:
@bkueng Just tested #15867 - it fixes the problem.

closed 2020-10-03 by dagar