PX4/PX4-Autopilot #15832: Simulator crashes after running make px4_sitl jmavsim

AdamBCo, 2020-09-27:
**Describe the bug**
The simulator crashes when running `make px4_sitl_default jmavsim`.

This is what I am seeing in my console:

```
BUILD SUCCESSFUL
Total time: 0 seconds
Options parsed, starting Sim.
Starting GUI...
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by javax.media.j3d.JoglPipeline (rsrc:j3dcore.jar) to method sun.awt.AppContext.getAppContext()
WARNING: Please consider reporting this to the maintainers of javax.media.j3d.JoglPipeline
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
INFO  [simulator] Simulator connected on TCP port 4560.
Init MAVLink
INFO  [commander] LED: open /dev/led0 failed (22)
/Users/adamcooper/Firmware/build/px4_sitl_default/tmp/rootfs/etc/init.d/rc.vehicle_setup: line 81: /etc/init.d/rc.interface: No such file or directory
ERROR [px4] Startup script returned with return value: 256
```
**To Reproduce**
Steps to reproduce the behavior:
1. Follow the instructions for Mac installation found [here](https://dev.px4.io/master/en/setup/dev_env_mac.html)
2. Follow the instructions for Building the code found [here](https://dev.px4.io/master/en/setup/building_px4.html)
3. Navigate into the Firmware directory and start jMAVSim using the following command: `make px4_sitl jmavsim`

**Expected behavior**
The simulator should not crash

**Log Files and Screenshots**

```
BUILD SUCCESSFUL
Total time: 0 seconds
Options parsed, starting Sim.
Starting GUI...
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by javax.media.j3d.JoglPipeline (rsrc:j3dcore.jar) to method sun.awt.AppContext.getAppContext()
WARNING: Please consider reporting this to the maintainers of javax.media.j3d.JoglPipeline
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
INFO  [simulator] Simulator connected on TCP port 4560.
Init MAVLink
INFO  [commander] LED: open /dev/led0 failed (22)
/Users/adamcooper/Firmware/build/px4_sitl_default/tmp/rootfs/etc/init.d/rc.vehicle_setup: line 81: /etc/init.d/rc.interface: No such file or directory
ERROR [px4] Startup script returned with return value: 256
```

julianoes, 2020-09-28:
I think this is a duplicate of #15795.

Jaeyoung-Lim, 2020-12-07:
Closing as a duplicate of https://github.com/PX4/PX4-Autopilot/pull/15970

closed 2020-12-07 by Jaeyoung-Lim