PX4/PX4-Autopilot #15834: Error while 'make px4_sitl_default gazebo'

bamboosdu, 2020-09-28:
Ubuntu18.04
Gazebo9.0
ROS:melodic
After I set up all the thing, I met the problem about px4_sitl_default.
`(base) zq@zq:~/src/Firmware$ make px4_sitl_default gazebo
[2/6] Performing configure step for 'sitl_gazebo'
-- install-prefix: /usr/local
-- Using C++17 compiler
-- Found Boost: /usr/local/include (found suitable version "1.65.1", minimum required is "1.58") found components: system thread filesystem chrono date_time atomic 
CMake Warning (dev) at /home/zq/cmake/share/cmake-3.18/Modules/FindPackageHandleStandardArgs.cmake:273 (message):
  The package name passed to `find_package_handle_standard_args` (PkgConfig)
  does not match the name of the calling package (gazebo).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  /home/zq/cmake/share/cmake-3.18/Modules/FindPkgConfig.cmake:59 (find_package_handle_standard_args)
  /usr/lib/x86_64-linux-gnu/cmake/gazebo/gazebo-config.cmake:30 (include)
  CMakeLists.txt:34 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found Boost: /usr/local/include (found suitable version "1.65.1", minimum required is "1.40.0") found components: thread system filesystem program_options regex iostreams date_time chrono atomic 
-- Found Boost: /usr/local/include (found version "1.65.1")  
-- Looking for OGRE...
-- Found Ogre Ghadamon (1.9.0)
-- Looking for OGRE_Paging...
-- Found OGRE_Paging: optimized;/usr/lib/x86_64-linux-gnu/libOgrePaging.so;debug;/usr/lib/x86_64-linux-gnu/libOgrePaging.so
-- Looking for OGRE_Terrain...
-- Found OGRE_Terrain: optimized;/usr/lib/x86_64-linux-gnu/libOgreTerrain.so;debug;/usr/lib/x86_64-linux-gnu/libOgreTerrain.so
-- Looking for OGRE_Property...
-- Found OGRE_Property: optimized;/usr/lib/x86_64-linux-gnu/libOgreProperty.so;debug;/usr/lib/x86_64-linux-gnu/libOgreProperty.so
-- Looking for OGRE_RTShaderSystem...
-- Found OGRE_RTShaderSystem: optimized;/usr/lib/x86_64-linux-gnu/libOgreRTShaderSystem.so;debug;/usr/lib/x86_64-linux-gnu/libOgreRTShaderSystem.so
-- Looking for OGRE_Volume...
-- Found OGRE_Volume: optimized;/usr/lib/x86_64-linux-gnu/libOgreVolume.so;debug;/usr/lib/x86_64-linux-gnu/libOgreVolume.so
-- Looking for OGRE_Overlay...
-- Found OGRE_Overlay: optimized;/usr/lib/x86_64-linux-gnu/libOgreOverlay.so;debug;/usr/lib/x86_64-linux-gnu/libOgreOverlay.so
-- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so;-lpthread (found suitable version "3.0.0", minimum required is "2.3.0") 
-- Checking for module 'uuid'
--   Found uuid, version 2.31.1
-- Found UUID: TRUE  
-- Checking for module 'tinyxml2'
--   Found tinyxml2, version 6.0.0
-- Looking for dlfcn.h - found
-- Looking for libdl - found
-- FreeImage.pc not found, we will search for FreeImage_INCLUDE_DIRS and FreeImage_LIBRARIES
-- Found UUID: TRUE  
-- Checking for module 'gts'
--   Found gts, version 0.7.6
-- Checking for module 'libswscale'
--   Found libswscale, version 4.8.100
-- Checking for module 'libavdevice >= 56.4.100'
--   Found libavdevice , version 57.10.100
-- Checking for module 'libavformat'
--   Found libavformat, version 57.83.100
-- Checking for module 'libavcodec'
--   Found libavcodec, version 57.107.100
-- Checking for module 'libavutil'
--   Found libavutil, version 55.78.100
-- Checking for module 'jsoncpp'
--   Found jsoncpp, version 1.7.4
-- Checking for module 'yaml-0.1'
--   Found yaml-0.1, version 0.1.7
-- Checking for module 'libzip'
--   Found libzip, version 1.1.2
-- Building klt_feature_tracker without catkin
-- Building OpticalFlow with OpenCV
-- catkin DISABLED
-- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so;-lpthread (found version "3.0.0") 
Gazebo version: 9.14
-- Configuring done
-- Generating done
-- Build files have been written to: /home/zq/src/Firmware/build/px4_sitl_default/build_gazebo
[3/6] Performing build step for 'sitl_gazebo'
FAILED: external/Stamp/sitl_gazebo/sitl_gazebo-build 
cd /home/zq/src/Firmware/build/px4_sitl_default/build_gazebo && /home/zq/cmake/bin/cmake --build . && /home/zq/cmake/bin/cmake -E touch /home/zq/src/Firmware/build/px4_sitl_default/external/Stamp/sitl_gazebo/sitl_gazebo-build
[1/2] Building CXX object CMakeFiles/gazebo_opticalflow_plugin.dir/src/gazebo_opticalflow_plugin.cpp.o
FAILED: CMakeFiles/gazebo_opticalflow_plugin.dir/src/gazebo_opticalflow_plugin.cpp.o 
/usr/bin/c++ -DLIBBULLET_VERSION=2.87 -DLIBBULLET_VERSION_GT_282 -Dgazebo_opticalflow_plugin_EXPORTS -I/home/zq/src/Firmware/Tools/sitl_gazebo/include -I. -I/usr/include/eigen3 -I/usr/include/eigen3/eigen3 -I/usr/include/gazebo-9/gazebo/msgs -I/home/zq/src/Firmware/mavlink/include -I/usr/local/include/Paging -I/home/zq/src/Firmware/Tools/sitl_gazebo/external/OpticalFlow/include -I/home/zq/src/Firmware/Tools/sitl_gazebo/external/OpticalFlow/external/klt_feature_tracker/include -isystem /usr/include/gazebo-9 -isystem /usr/include/bullet -isystem /usr/include/simbody -isystem /usr/include/sdformat-6.2 -isystem /usr/include/ignition/math4 -isystem /usr/include/OGRE -isystem /usr/include/OGRE/Terrain -isystem /usr/include/OGRE/Paging -isystem /usr/include/ignition/transport4 -isystem /usr/include/ignition/msgs1 -isystem /usr/include/ignition/common1 -isystem /usr/include/ignition/fuel_tools1 -isystem /usr/local/include/opencv4 -isystem /usr/include/uuid -Wno-deprecated-declarations -fPIC -I/usr/include/uuid -I/usr/include/x86_64-linux-gnu -std=gnu++1z -MD -MT CMakeFiles/gazebo_opticalflow_plugin.dir/src/gazebo_opticalflow_plugin.cpp.o -MF CMakeFiles/gazebo_opticalflow_plugin.dir/src/gazebo_opticalflow_plugin.cpp.o.d -o CMakeFiles/gazebo_opticalflow_plugin.dir/src/gazebo_opticalflow_plugin.cpp.o -c /home/zq/src/Firmware/Tools/sitl_gazebo/src/gazebo_opticalflow_plugin.cpp
In file included from /home/zq/src/Firmware/Tools/sitl_gazebo/src/gazebo_opticalflow_plugin.cpp:24:0:
/home/zq/src/Firmware/Tools/sitl_gazebo/src/gazebo_opticalflow_plugin.cpp: In member function ‘virtual void gazebo::OpticalFlowPlugin::Load(gazebo::sensors::SensorPtr, sdf::ElementPtr)’:
/home/zq/src/Firmware/Tools/sitl_gazebo/include/gazebo_opticalflow_plugin.h:43:18: error: ‘TRUE’ was not declared in this scope
 #define HAS_GYRO TRUE
                  ^
/home/zq/src/Firmware/Tools/sitl_gazebo/include/gazebo_opticalflow_plugin.h:43:18: note: in definition of macro ‘HAS_GYRO’
 #define HAS_GYRO TRUE
                  ^~~~
ninja: build stopped: subcommand failed.
ninja: build stopped: subcommand failed.
Makefile:193: recipe for target 'px4_sitl_default' failed
make: *** [px4_sitl_default] Error 1
`

Jaeyoung-Lim, 2020-09-28:
Which firmware are you using?

Jaeyoung-Lim, 2020-12-06:
Closing since the author didn't respond

closed 2020-12-06 by Jaeyoung-Lim