PX4/PX4-Autopilot #15835: Pixhawk cube + GPS Here modul, not detecting magnetometer

MartinStokelj, 2020-09-28:
**Describe the bug**
Cube does not calibrate external magnetometer on GPS Here module

**To Reproduce**
Steps to reproduce the behavior:
Connect GPS Here module and start compass calibration

**Expected behavior**
It should detect external magnetometer

**Log Files and Screenshots**
[image]
[putty.txt](https://github.com/PX4/Firmware/files/5290515/putty.txt)

**Drone (please complete the following information):**
- VTOL DetlaQuad

@dagar tagging you as you wanted

dagar, 2020-09-28:
Looks like it's the `icm20948` on I2C which has an ak09916 magnetometer. 

We have a driver for both the icm20948 and ak09916, so I believe the problem in this case is that the icm20948 needs to be configured in i2c passthrough mode so that the ak09916 is directly on available on the bus.

MartinStokelj, 2020-09-28:
So what can I do?

nsh> ak09916 status
INFO  [SPI_I2C] Not running
nsh> icm20948 status
INFO  [SPI_I2C] Not running
nsh>

dagar, 2020-09-28:
2 Options
1) extend the existing icm20948 to work on I2C
     - at the moment I'd say we actually don't want this because that extra external IMU on I2C probably isn't usable for flight and we don't currently have a mechanism to treat it differently
2) create a very simple icm20948_i2c that simply places the device in I2C passthrough mode so that the ak09916 is available directly on the I2C bus, then start the standalone ak09916 driver.

MartinStokelj, 2020-09-28:
According to https://docs.px4.io/master/en/assembly/quick_start_cube.html
The recommended GPS modules are the Here and Here+, both of which incorporate a GPS module, Compass, Safety Switch and LEDs.
But it looks like this do not work anymore without modifications?

I have also DroTek (Ublox GPS + compass) which works fine and is correctly detected with PX4, should I use this one instead of Here module?

dagar, 2020-09-30:
> I have also DroTek (Ublox GPS + compass) which works fine and is correctly detected with PX4, should I use this one instead of Here module?

Yes, the vast majority of these GPS modules are all the same ublox m8n.

I'll look into fixing the icm20948.

MartinStokelj, 2020-10-02:
So even DroTek  didn't work. It has found the external magnetometer but the GPS was not working. I have ordered Here2 and now both borks, GPS and external mag on I2C.

closed 2020-10-02 by MartinStokelj

dagar, 2020-10-02:
Thanks for the update, I'll try to have the icm20948 on i2c fully working soon.