PX4/PX4-Autopilot #15839: Nuttx 9.1.0+ Issues

davids5, 2020-09-28:
ver all reports  OS version: Release 8.2.0 (134349055)

dagar, 2020-09-28:
`nxp_fmuk66-e_socketcan` build is broken

`NuttX/nuttx/arch/arm/src/chip/kinetis_flexcan.c:645: undefined reference to `clock_systimespec'`

dagar, 2020-09-28:
Possible memory corruption somewhere.
 - log_writer_file stack overflow, but after increase the thread reported even less stack usage https://github.com/PX4/Firmware/pull/15765
 - similar case for mavlink https://github.com/PX4/Firmware/pull/15821

davids5, 2020-09-28:
> Possible memory corruption somewhere.
> 
> * log_writer_file stack overflow, but after increase the thread reported even less stack usage #15765
> * similar case for mavlink #15821

This one may be the TLS/change effecting coloring the stack there is a PR in upstream in progress....

dagar, 2020-09-28:
### nxp_fmuk66-v3_default
 - hard faulting when running `tests` http://ci.px4.io:8080/blue/organizations/jenkins/PX4_misc%2FFirmware-hardware/detail/master/2338/pipeline/1761

### px4_fmu-v3_default
 - intermittant hard fault in `tests` http://ci.px4.io:8080/blue/organizations/jenkins/PX4_misc%2FFirmware-hardware/detail/master/2337/pipeline/1140
 - seems to be getting caught in boot loop dumping hard fault

dagar, 2020-09-28:
Hard fault dump is stopping here (!), then rebooting.

``` Console

System fault Occurred on: 2020-09-28-18:21:22
 Type:Hard Fault in file:armv7-m/arm_hardfault.c at line: 134 running task: tests
 FW git-hash: 3bef29cbf1c1ac6b717f8ccf0090a017dc863d32
 Build datetime: Sep 28 2020 14:32:31
 Build url: localhost 
 Processor registers: from 0x2002fbc4
 r0:0x00000000 r1:0x2002c480  r2:0x00000000  r3:0x00000000  r4:0x00000000  r5:0x0000000c r6:0x0000002e r7:0x0816d65e
 r8:0x0816d70c r9:0x00000002 r10:0x2002fd00 r11:0x0816d868 r12:0x00000000  sp:0x2002fc98 lr:0x080e1ce5 pc:0x00000000
 xpsr:0x60000000 basepri:0xarm_hardfault: Hard Fault:
arm_hardfault:   IRQ: 3 regs: 0x10002a6c
arm_hardfault:   BASEPRI: 000000f0 PRIMASK: 00000000 IPSR: 00000003 CONTROL: 00000000
arm_hardfault:   CFAULTS: 00008200 HFAULTS: 40000000 DFAULTS: 00000000 BFAULTADDR: 00fffe00 AFAULTS: 00000000
arm_hardfault: PANIC!!! Hard fault: 40000000
up_assert: Assertion failed at file:armv7-m/arm_hardfault.c line: 134 task: init
up_registerdump: R0: 00fffe00 00000200 deadbeef 00000080 0000002a 10003d40 10003f18 000000c8
up_registerdump: R8: 20002500 00000001 00000000 00000001 00000000 10002b40 080cc6c7 0800ddd6
up_registerdump: xPSR: 01000000 BASEPRI: 000000f0 CONTROL: 00000000
up_registerdump: EXC_RETURN: ffffffe9
up_dumpstate: sp:     20002570
up_dumpstate: IRQ stack:
up_dumpstate:   base: 20002600
up_dumpstate:   size: 00000200
up_dumpstate:   used: 00000188
up_stackdump: 20002560: 20002570 20002600 00000b70 0800d825 000000f0 00000000 00000001 00000000
up_stackdump: 20002580: 10002b40 080cc6c7 0800ddd6 e000ed2c 00fffe00 e000ed2c 00fffe00 00000000
up_stackdump: 200025a0: 00000000 20002500 00000001 00000000 00000001 08005aa7 08006815 080057df
up_stackdump: 200025c0: 40000000 00000000 00fffe00 00000000 20002600 00000003 00000003 0800702b
up_stackdump: 200025e0: 000000f0 0800585d 000000f0 10002a6c 10003d40 10003f18 000000c8 08004fc7
up_dumpstate: sp:     10002b40
up_dumpstate: User stack:
up_dumpstate:   base: 10002df8
up_dumpstate:   size: 00000b70
up_dumpstate:   used: 00000404
up_stackdump: 10002b40: 0816827c 00000000 00000000 00000000 00000000 00000000 00000000 00000000
up_stackdump: 10002b60: 00000000 00000000 00000000 00000000 00000000 00000000 00000002 10001dd4
up_stackdump: 10002b80: 20002824 20007a48 00000000 00000200 20002500 01000000 00000000 00000000
up_stackdump: 10002ba0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
up_stackdump: 10002bc0: 00000000 00000000 00000000 00000000 00000080 08004659 20000100 08004807
up_stackdump: 10002be0: 080047c1 20000100 0000003f 08005dbd 00000000 00000000 00000000 2000015c
up_stackdump: 10002c00: 00000000 0800744b 00000065 20000100 00000065 08006315 10003d40 10003d40
up_stackdump: 10002c20: 00000001 00000075 00000001 0815c414 00000000 08012edf 00000001 00000065
up_stackdump: 10002c40: 00000065 10001f54 00000065 08012ef5 00000065 080cc95d 40000000 00000000
up_stackdump: 10002c60: e000ed34 e000ed38 00000000 080e1ce5 00000000 00000010 73666320 78303a72
up_stackdump: 10002c80: 78383025 73666820 78303a72 78383025 66642020 303a7273 38302578 6d202078
up_stackdump: 10002ca0: 7273666d 2578303a 20783830 73666220 78303a72 78383025 73666120 78303a72
up_stackdump: 10002cc0: e0000000 00000003 10003e10 10003d40 00000001 0815c414 00000000 080cd05d
up_stackdump: 10002ce0: 000000c8 00000001 00000307 00000158 00000000 0c690401 5f7229a2 2441fca3
up_stackdump: 10002d00: 00000002 00002158 00000000 00000002 00000001 00000010 00000000 0001001c
up_stackdump: 10002d20: 00000000 0812bec1 10003bc0 00000000 00000004 00000180 00000180 00000040
up_stackdump: 10002d40: ffffffff 00000000 081955d8 08004d91 2000658c 0002001c 20000d30 00000010
up_stackdump: 10002d60: 00000000 0812b753 000003e8 00000000 08004d91 00000000 00000000 000003e8
up_stackdump: 10002d80: 00000000 000f4240 00000000 00000000 10002df8 00000000 00000000 00000000
up_stackdump: 10002da0: 00000000 00000000 00000000 08011407 0000ff01 00000000 00000001 080091ff
up_stackdump: 10002dc0: 00000000 080088c9 00000000 00000064 00000000 20002ff0 080091d9 0800f675
up_stackdump: 10002de0: 080091d9 00000001 00000000 08007f4b 00000000 00000000 10002e00 00000000
up_taskdump: Idle Task: PID=0 Stack Used=0 of 0
up_taskdump: hpwork: PID=1 Stack Used=380 of 1264
up_taskdump: lpwork: PID=2 Stack Used=340 of 1616
up_taskdump: init: PID=3 Stack Used=1028 of 2928
up_taskdump: wq:manager: PID=4 Stack Used=340 of 1256
!
```

davids5, 2020-10-14:
All issues have been addressed once https://github.com/PX4/Firmware/pull/15960 is in this can be closed

closed 2020-10-14 by dagar