PX4/PX4-Autopilot #15860: INA226 internal register configuration, overriden by param?

roangel, 2020-10-01:
https://github.com/PX4/Firmware/blob/c1936dab1fb511a34b5bb0c6b51abd537c96cc84/src/drivers/power_monitor/ina226/ina226.cpp#L70-L76

I had to debug very thoroughly until I realised that all my configuration written in the .h file, all along, was being overwritten by a parameter. Not sure if it's the best idea to have a parameter for internal register configuration? Maybe for debugging is ok, but for stable code?

dagar, 2020-10-01:
At a minimum I would suggest breaking it apart into separate parameters for each piece of INA226_CONFIG that actually needs to be user configurable.

``` C
#define INA226_CONFIG (INA226_MODE_SHUNT_BUS_CONT | INA226_VSHCT_588US | INA226_VBUSCT_588US | INA226_AVERAGES_64)
```

roangel, 2020-10-01:
> At a minimum I would suggest breaking it apart into separate parameters for each piece of INA226_CONFIG that actually needs to be user configurable.
> 
> ```c
> #define INA226_CONFIG (INA226_MODE_SHUNT_BUS_CONT | INA226_VSHCT_588US | INA226_VBUSCT_588US | INA226_AVERAGES_64)
> ```

Yeah, true as well. However, personally here I would just remove all params, it's not like people change a shunt resistor every week, and I think we are used to configuring this driver stuff in the header file. I didn't even think about params for these until I realised

closed 2026-04-10 by github-actions[bot]