PX4/PX4-Autopilot #15874: mavlink MANUAL_CONTROL review

dagar, 2020-10-03:
[image]

 - Z range (throttle)
     - stick throttle range [-1, 1] -> [0, 1] (multicopter, fixedwing)
     - it seems like handling this firmware side would be cleaner and give us more flexibility
 - `buttons` bitfield
     - currently unused except when emulating RC
 - COM_RC_IN_MODE
     - 0 RC Transmitter
     - 1 Joystick/No RC Checks
          - simply skips arming/disarming via sticks and a few manual control checks, 
     - 2 Virtual RC by Joystick
          - publishes fake raw RC (input_rc) from mavlink MANUAL_CONTROL
          - should be removed given we the existance of RC_CHANNELS_OVERRIDE (https://mavlink.io/en/messages/common.html#RC_CHANNELS_OVERRIDE)

dagar, 2020-10-03:
Other ideas
 - on a gamepad its easy to unintentionally yaw when pulling the throttle all the way down to land
     - dynamic yaw deadzone?
 - mavlink receiver thread special handling if receiving MANUAL_CONTROL?
     - priority boost?

MaEtUgR, 2020-10-07:
> stick throttle range [-1, 1] -> [0, 1] (multicopter, fixedwing)

I'd have all axes in the same range [-1,1]. That would make it much more consistent and avoid unnecessary extra handling. https://github.com/PX4/Firmware/issues/9331

13022591351, 2024-07-26:
> [image]
> 
> * Z 范围（油门）
>   
>   * 摇杆油门范围 [-1, 1] -> [0, 1] （多旋翼飞行器，固定翼）
>   * 处理这个固件方面似乎会更清晰，并且让我们在车型之间有更大的灵活性
> * `buttons`位域
>   
>   * 目前不使用，除非模拟 RC
>   * 可以在 mavlink 级别配置，还是我们添加一个更通用的机制
> * COM_RC_IN_MODE
>   
>   * 0 遥控发射器
>   * 1 个操纵杆/无 RC 检查
>     
>     * 只需跳过通过操纵杆和一些手动控制检查进行布防/撤防即可，
>   * 2 通过操纵杆进行虚拟遥控
>     
>     * 从 mavlink MANUAL_CONTROL 发布伪造的原始 RC（input_rc）
>     * 鉴于 RC_CHANNELS_OVERRIDE 的存在，应该将其删除（https://mavlink.io/en/messages/common.html#RC_CHANNELS_OVERRIDE）

I think MANUAL_CONTROL cannot completely replace the function of RC_CHANNELS_OVERRIDE. I recently wanted to develop a device based on Microhard HP900 to implement TEL RTCM RC_IN through a single UART interface.
After checking the Mavlink protocol, I found that RC_CHANNELS_OVERRIDE matches my needs very well.
Maybe my project is mainly used in fixed-wing aircraft, and needs signals other than X, Y, Z, and R, such as flap control, landing gear retraction, etc.