PX4/PX4-Autopilot #15891: How to send values from PX4 to VESC without Remote Input, How to see that values at VESC side, maybe on terminal.c

sam-rke, 2020-10-06:
Hi there,
I need to use canbus on PX4 for that i need to send some hardcoded data from px4  to VESC6 over canbus using UAVCAN. Now i am not sure about:
1. In which files the change can be done.
2. where to see that reflected change on VESC as for now i have no motors, so is it possible to trace out those changes using some print statement in File terminal.c in vesc firmware?

Any suggestions for above problems?

RicardoM17, 2020-10-12:
@sam-rke we didn't have to make any changes to PX4 side in order to fly with VESC (though there were some improvements).

Just setup the Pixhawk to fly with UAVCAN and same with the VESC. make sure the baud rates match etc.

Not sure if that answers your questions.

sam-rke, 2020-10-12:
> @sam-rke we didn't have to make any changes to PX4 side in order to fly with VESC (though there were some improvements).
> 
> Just setup the Pixhawk to fly with UAVCAN and same with the VESC. make sure the baud rates match etc.
> 
> Not sure if that answers your questions.
> 
> If you feel like it would be useful I can add some documentation on how to setup VESC, but I don't have much time to test atm.

Hi @RicardoM17 Thank-you for your response.
Actually i am just trying to understand the UAVCAN packet sending mechanism that's why I'm trying to send anything(maybe just simple character) from PX4 to VESC over CAN bus.
So my question is where I can find the code in which I can send CAN packet by making some changes, I tried implementing uORB publisher and subscriber in Hello example and it's working fine as i saw updated values on QGC console, Same thing i wanna do with UAVCAN?

RicardoM17, 2020-10-12:
@sam-rke that makes sense. I'm actually not sure where that is located. I'll be also eavesdroping in that cause as I'd like to implement current control with UAVCAN + PX4 + VESC, which I understand is currently not the case.

Perhaps @pavel-kirienko you can hep out here and answer @sam-rke 's questions?

ryanjAA, 2020-10-15:
@RicardoM17 if you have a vesc working with px4 definitely document it so it can be added if someone wants to try. At the very least it can then be listed as compatible.

RicardoM17, 2020-10-15:
@ryanjAA yes I went ahead and made this PR yesterday which was already merged https://github.com/PX4/px4_user_guide/pull/831

sam-rke, 2020-10-15:
Okay @RicardoM17  and @ryanjAA  thankyou guys for you time.
I am wating for my VESC to be delivered as last one is not working. It got bricked everytime I flashed modified Firmware into it, after that i have to flash Bootloader. But after some Flashing maybe 5-6 time it stopped getting detected in vesc tool.
Now I'll be working with next one so I wanna be more careful this time.

RicardoM17, 2020-10-15:
@sam-rke are you aware that you must wait at least 10sec after flashing the VESC firmware before unplugging it? if you don't do it then you risk bricking your VESC. If you are bricking your VESC so often maybe you are missing this step.

sam-rke, 2020-10-15:
> @sam-rke are you aware that you must wait at least 10sec after flashing the VESC firmware before unplugging it? if you don't do it then you risk bricking your VESC. If you are bricking your VESC so often maybe you are missing this step.

Yes @RicardoM17  I knew that, It's just if I am flashing Firmware without modification it never got bricked but if i change even one line in Firmware and Flashed it , VESC got bricked.

closed 2026-04-10 by github-actions[bot]