PX4/PX4-Autopilot #15894: SITL test failures to investigate

dagar, 2020-10-06:
- https://github.com/PX4/Firmware/pull/15892/checks?check_run_id=1215051101

dagar, 2020-10-06:
[image]

Jaeyoung-Lim, 2020-10-06:
The failure seems intermittent: re-running the tests can make the tests fail / pass on different tests

dagar, 2020-10-06:
> The failure seems intermittent: re-running the tests can make the tests fail / pass on different tests

"Have you trying turning it off and on again?"

dagar, 2020-10-06:
We should fix the ulg upload in these.
[image]

Jaeyoung-Lim, 2020-10-06:
@dagar What do you mean by off/on? - I run the same tests in sitl_gazebo, and running it 3-4 times make it pass.

- We need to investigate this, but not sure where to start

dagar, 2020-10-06:
@julianoes what do you think about extending the mavsdk test runner (python) to also upload logs to flight review (with all metadata)? I think it would help debug failures if we could trivially compare flight review runs side-by-side (failure and last success).

closed 2020-11-16 by dagar