PX4/PX4-Autopilot #15901: Jammed while installing toolchains; sourcing 'ubuntu.sh'

darkasassin44, 2020-10-07:
**Describe the bug**
Ubuntu 18.04 LTS

while I'm installing toolchains, has jammed at here -> https://dev.px4.io/v1.10/en/setup/dev_env_linux_ubuntu.html#sim_nuttx
this is the current terminal window.
```
Installing PX4 simulation dependencies
Reading package lists...
Building dependency tree...
Reading state information...
ant is already the newest version (1.10.5-3~18.04).
openjdk-8-jdk is already the newest version (8u265-b01-0ubuntu2~18.04).
openjdk-8-jre is already the newest version (8u265-b01-0ubuntu2~18.04).
The following packages were automatically installed and are no longer required:
  efibootmgr libfwup1 libllvm9 libmbim-glib4 libmbim-proxy libqmi-glib5
  libqmi-proxy usb-modeswitch usb-modeswitch-data
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 3 not upgraded.
--2020-10-07 14:48:19--  http://packages.osrfoundation.org/gazebo.key
Resolving packages.osrfoundation.org (packages.osrfoundation.org)... 52.52.171.73
Connecting to packages.osrfoundation.org (packages.osrfoundation.org)|52.52.171.73|:80... connected.
HTTP request sent, awaiting response... No data received.
Retrying.

--2020-10-07 14:49:20--  (try: 2)  http://packages.osrfoundation.org/gazebo.key
Connecting to packages.osrfoundation.org (packages.osrfoundation.org)|52.52.171.73|:80... connected.
HTTP request sent, awaiting response... No data received.
Retrying.

--2020-10-07 14:50:22--  (try: 3)  http://packages.osrfoundation.org/gazebo.key
Connecting to packages.osrfoundation.org (packages.osrfoundation.org)|52.52.171.73|:80... connected.
HTTP request sent, awaiting response... No data received.
Retrying.

--2020-10-07 14:51:26--  (try: 4)  http://packages.osrfoundation.org/gazebo.key
Connecting to packages.osrfoundation.org (packages.osrfoundation.org)|52.52.171.73|:80... connected.
HTTP request sent, awaiting response... 
```

both v1.10.0 and v1.11.0, same error has occurred.
Please tell me how to go through this stage.

darkasassin44, 2020-10-07:
I already installed ROS Melodic Desktop with official guide, and I'm planning to use SITL with Cube Black(Pixhawk 2.0)

hamishwillee, 2020-10-08:
@darkasassin44 
There are two scripts in https://dev.px4.io/v1.10/en/setup/dev_env_linux_ubuntu.html - one just for installing ROS and one for installing the other environments. They are both designed/tested to run on a clean system - ie not install over pre-existing setups. The reason is that ROS installs gazebo already and if you layer them over each other things can break.

So there is no "official" way to support this. 

I would try running this script: https://raw.githubusercontent.com/PX4/Devguide/v1.10.0/build_scripts/ubuntu_sim_ros_melodic.sh

If that doesn't work I'd try removing gazebo and trying it again. Sorry can't be more helpful.

closed 2021-01-17 by Jaeyoung-Lim