PX4/PX4-Autopilot #15902: ERROR [px4] Startup script returned with return value: 512

stijojoseph, 2020-10-07:
$ make px4_sitl jmavsim                                                                                                 -- PX4 version: v1.11.0-rc3-432-gc01fabaf11                                                                             -- PX4 config file: /cygdrive/c/SPB_Data/Firmware/boards/px4/sitl/default.cmake                                         -- PX4 config: px4_sitl_default                                                                                         -- PX4 platform: posix                                                                                                  -- PX4 lockstep: enabled                                                                                                -- cmake build type: RelWithDebInfo                                                                                     -- The CXX compiler identification is GNU 9.3.0                                                                         -- The C compiler identification is GNU 9.3.0                                                                           -- The ASM compiler identification is GNU                                                                               -- Found assembler: /usr/bin/cc                                                                                         -- Check for working CXX compiler: /usr/bin/c++.exe                                                                     -- Check for working CXX compiler: /usr/bin/c++.exe -- works                                                            -- Detecting CXX compiler ABI info                                                                                      -- Detecting CXX compiler ABI info - done                                                                               -- Detecting CXX compile features                                                                                       -- Detecting CXX compile features - done                                                                                -- Check for working C compiler: /usr/bin/cc                                                                            -- Check for working C compiler: /usr/bin/cc -- works                                                                   -- Detecting C compiler ABI info                                                                                        -- Detecting C compiler ABI info - done                                                                                 -- Detecting C compile features                                                                                         -- Detecting C compile features - done                                                                                  -- Building for code coverage                                                                                           -- Found PythonInterp: /usr/bin/python3 (found suitable version "3.6.9", minimum required is "3")                       -- build type is RelWithDebInfo                                                                                         -- PX4 ECL: Very lightweight Estimation & Control Library v1.9.0-rc1-450-g4c2355a                                       -- ROMFS: px4fmu_common                                                                                                 -- Configuring done                                                                                                     -- Generating done                                                                                                      -- Build files have been written to: /cygdrive/c/SPB_Data/Firmware/build/px4_sitl_default                               [0/759] git submodule src/drivers/gps/devices                                                                           [1/759] git submodule src/lib/ecl                                                                                       [2/759] git submodule mavlink/include/mavlink/v2.0                                                                      [6/759] git submodule Tools/jMAVSim                                                                                     [758/759] cd /cygdrive/c/SPB_Data/Firmware/build/px4_sitl_...mware /cygdrive/c/SPB_Data/Firmware/build/px4_sitl_default SITL ARGS                                                                                                               sitl_bin: /cygdrive/c/SPB_Data/Firmware/build/px4_sitl_default/bin/px4.exe                                              debugger: none                                                                                                          program: jmavsim                                                                                                        model: none                                                                                                             world: none                                                                                                             src_path: /cygdrive/c/SPB_Data/Firmware                                                                                 build_path: /cygdrive/c/SPB_Data/Firmware/build/px4_sitl_default                                                        empty model, setting iris as default                                                                                    SITL COMMAND: "/cygdrive/c/SPB_Data/Firmware/build/px4_sitl_default/bin/px4.exe" "/cygdrive/c/SPB_Data/Firmware/build/px4_sitl_default"/etc -s etc/init.d-posix/rcS -t "/cygdrive/c/SPB_Data/Firmware"/test_data                                INFO  [px4] Creating symlink /cygdrive/c/SPB_Data/Firmware/build/px4_sitl_default/etc -> /cygdrive/c/SPB_Data/Firmware/build/px4_sitl_default/tmp/rootfs/etc                                                                                                                                                                                                            ______  __   __    ___                                                                                                  | ___ \ \ \ / /   /   |                                                                                                 | |_/ /  \ V /   / /| |                                                                                                 |  __/   /   \  / /_| |                                                                                                 | |     / /^\ \ \___  |                                                                                                 \_|     \/   \/     |_/                                                                                                                                                                                                                         px4 starting.                                                                                                                                                                                                                                   INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0                                                      /cygdrive/c/SPB_Data/Firmware/build/px4_sitl_default/bin/px4-alias.sh: line 42: `source': is a special builtin          ERROR [px4] Startup script returned with return value: 512

Jaeyoung-Lim, 2020-10-09:
@stijojoseph Could you be more descriptive on what you were trying to do, and what is not working for you?

xdwgood, 2020-10-10:
I get same issue. when I try run `make px4_sitl jmavsim` on my windows10

stijojoseph, 2020-10-10:
@Jaeyoung-Lim @xdwgood 
issue was solved same day itself

Firmware/platforms/posix/src/px4/common/px4-alias.sh_in

Line 31 in a244883
replace 'source' with 'rs'
 source() { 
gets rid of the error 512 'source': is a special builtin but then it fails because of rc.vehicle_setup invoking NuttX path scripts -> No such file or directory e.g. with multicopter here:
Firmware/ROMFS/px4fmu_common/init.d/rc.vehicle_setup

replace 'source' with 'rs'
Lines 80 to 84 in a244883

 # Load mixer and configure outputs. 
 source /etc/init.d/rc.interface 
  
 # Start standard multicopter apps. 
 source /etc/init.d/rc.mc_apps

closed 2020-10-10 by stijojoseph