PX4/PX4-Autopilot #15907: mRo v2.1 and Taoglas GPS

netkimjh, 2020-10-07:
Hi,

I am struggling for integrating Taoglas GPS with mRo v2.1.
I confirmed the GPS is powered and working well with Ardupilot. 
But the problem is Taoglas GPS is not powered with any PX4 version.
The GPS is connected GPS/I2C port (UART 4, Serial 3)

FYI,
[mRo v2.1](https://docs.px4.io/master/en/flight_controller/mro_x2.1.html)
[Taoglas GPS](https://store.mrobotics.io/ProductDetails.asp?ProductCode=mro-gps003-mr)

Do you have any solution for this matter?

LorenzMeier, 2020-10-07:
There is no reason why it should not work.

@bkueng I have ordered the unit.

dagar, 2020-10-07:
@netkimjh I've used just about every version of every GPS from mRo with PX4 and I'm not aware of any current problems. You can see an old subset here https://github.com/PX4/Firmware/pull/15120.

Which version of PX4 exactly? Can you check the back of you mRo GPS to see if it's an m8n or m9n?

netkimjh, 2020-10-08:
@LorenzMeier and @dagar ,

I am using mRo v2.1 FCU. #15120 seems like not using mRo V2.1

1. Flashed and tested PX4 version from 1.9.0 to 1.11.1. Tried mro_x21_default.px4, auav_x21_default.px4, and px4_fmu-v2_default.px4.
(I downgraded the firmware from 1.11 due to crash of BAT parameter with QGC.)
2. NEO-M8N-0-10

I cannot see any light on GPS with PX4, so I checked the voltage of the port and confirmed that it outputs 5v.

bkueng, 2020-10-08:
@netkimjh what's the output of `gps status`?

dagar, 2020-10-08:
> I was not able to see any light on GPS with PX4, so I checked the voltage of the port and confirmed that it outputs 5v.

I'm not wondering if you're referring to the RGB LED on top of the unit rather than the GPS itself. It looks like the ncp5623c has another possible I2C address. Can you try https://github.com/PX4/Firmware/pull/15918?

netkimjh, 2020-10-09:
@bkueng @dagar ,
I am sorry that I cannot check it this week.
I will check and let you know next week.
Thanks.

bkueng, 2020-10-13:
I received the unit, it's indeed just the LED startup that's missing: `rgbled_ncp5623c start -X -q -a 0x38`

netkimjh, 2020-10-20:
@bkueng Hey, I meant there is not only no RGB LED, also not working gps. In QGC, I cannot see any GPS data with it. What should I try?

@dagar is there any update?

bkueng, 2020-10-21:
> @bkueng Hey, I meant there is not only no RGB LED, also not working gps. In QGC, I cannot see any GPS data with it. What should I try?

As I wrote above, please provide the output of `gps status`. See https://docs.qgroundcontrol.com/master/en/analyze_view/mavlink_console.html for how to access the shell.

netkimjh, 2020-10-21:
@bkueng 
I understand, and I received the following result.

[image]

So, it was working, and there was no RGB LED.. and I have "No GPS Lock for Vehicle"..

Thank you for clarifying this issue.

bkueng, 2020-10-22:
Correct, this looks all good.

closed 2020-10-22 by netkimjh

netkimjh, 2020-10-23:
@bkueng 
OK.
So far, I tried Firmware version 1.9.0, but it seems like not support mRo x2.1. So now I am using mro_x21_default v1.11.1.
However, it seems like esc calibration has an issue, correct?
I tried to use the esc calbiration in QGC, but there is an error that "Timeout waiting for battery".
 
Could you confirm it for me?

Thank you.