PX4/PX4-Autopilot #15909: Has PX4 firmware fully supported CUAV X7Pro yet?

saengphet, 2020-10-08:
Hello, I’m planning to use CUAV X7Pro which integrated the industrial-grade IMU “ADIS16470” and compass “rm3100”. I found that AUX ports (FMU) are not supported (#15110) for now as well as IMU ADIS16470 (look at comment out a line rc.board_sensors#L8 and spi.cpp#L40). Have anyone experienced using CUAV X7Pro before? How about the progress of X7Pro for using PX4?

**References**
https://github.com/PX4/Firmware/blob/master/boards/cuav/x7pro/init/rc.board_sensors#L8
https://github.com/PX4/Firmware/blob/master/boards/cuav/x7pro/src/spi.cpp#L40
https://github.com/PX4/Firmware/pull/15110
http://doc.cuav.net/flight-controller/x7/en/firmware.html#px4-firmware

**My post px4 discuss**
https://discuss.px4.io/t/has-px4-firmware-fully-supported-cuav-x7pro/18910

saengphet, 2020-10-12:
Hello, Do you have any information or recommendation for this question? Our team has already bought X7Pro and already tested on Ardupilot flight stack. However. I want to use PX4 because I have implemented/modified many features to meet the customer requirements. The one and most important feature are better IMU than other pixhawk or related platforms. @dagar @cuhome  @CUAVcaijie

saengphet, 2020-11-24:
Is any update for X7 pro especially using ADIS-IMU and AUX channel? @cuhome @CUAVcaijie @gitfishup

LorenzMeier, 2021-03-11:
Support will be part of the upcoming release.

closed 2021-03-11 by LorenzMeier

higwak, 2022-01-08:
Anyone wants to know about reality of cuav x7pro , let me know. 
My drone flew with this FC just for trial.

james-avetics, 2023-07-06:
> Anyone wants to know about reality of cuav x7pro , let me know. My drone flew with this FC just for trial.

Can you give me some insight?
What works and what doesn't work, especially the PWM output? Does all 14 ch works now?
How's the vibration isolation?

higwak, 2023-07-06:
Hi.
I bought x7pro and P8 Radio telemetry  couple of years ago.
None of them were working. No support from manufacturer because I didn't
bought them directly from cuav shop.